eclipse-milo/milo #199: Using enum for Method Arguments

hakjac, 2018-01-17:
Hi!

I'm implementing a OpcServer and like to use enums for some method arguments. I cannot find any example of this. I have successfully implemented methods with arguments of other types, such as UShort, Double etc. Can you give me some input or an example on how to use enums instead?

kevinherron, 2018-01-17:
Is this for an enum you want to define (another question in its own, only recently possible) or for an existing enum?

hakjac, 2018-01-18:
Hi Kevin! I want to define my own enums and use for input/output method arguments. I'm currently using numbers according to style in the SqrtMethod example.

E.g I want to accomplish something like this:

Enum: TempSensor (WaterTempSensor=0, AirTempSensor=1)
Enum ReturnCode (OK=0, NotOK=1)

Method: GetTemperature with
InputArgument: sensor, type:Sensor
OutputArgument: temperature, type Double
OutputArgument: returnCode, type: ReturnCode

Best Regards / Håkan

CB-IT, 2018-12-01:
Hi Håkan,
could you solve your problem? At the moment I'm facing a similar issue:
[https://stackoverflow.com/q/53490969](url)
Thanks in advance

kevinherron, 2019-02-28:
#415 introduces `AbstractMethodInvocationHandler` which should easily support enum argument types.

closed 2019-02-28 by kevinherron