eclipse-milo/milo #208: Client/Stack appends a forward slash to endpoint URLs

mregen, 2018-01-25:
Milo server doesn't accept appended slash, e.g. `opc.tcp://192.168.10.17:12685/`

https://github.com/OPCFoundation/UA-.NET/issues/35

Same issue in https://github.com/OPCFoundation/UA-.NETStandard, UA client trying to connect to the above server.

closed 2018-01-25 by kevinherron

jsebes1, 2021-03-01:
Still the issue is there with C# , any plan for  correcting it

kevinherron, 2021-03-01:
I thought this was fixed 3 years ago. 

Can you provide an example, logs, or a Wireshark capture that demonstrates this? Maybe it's being rejected in a different place than before...

jsebes1, 2021-03-04:
kevin , please see the  example below

for example.

from the sample client of OPCUA foundation
[(https://github.com/OPCFoundation/UA-.NETStandard)]

In ..\UA-.NETStandard-master\Applications\ConsoleReferenceClient\UAClient.cs
keep
private string m_serverUrl = "opc.tcp://x.y.z.a:4840"

in wire shark we can see that / is added at the end by default for url  "opc.tcp://x.y.z.a:4840/" 

problem seems to be in

```
public static EndpointDescription SelectEndpoint(
            string discoveryUrl,
            bool useSecurity,
            int discoverTimeout
            )
        {
            **var url = GetDiscoveryUrl(discoveryUrl);**
            var endpointConfiguration = EndpointConfiguration.Create();
            endpointConfiguration.OperationTimeout = discoverTimeout;

            // Connect to the server's discovery endpoint and find the available configuration.
            using (var client = DiscoveryClient.Create(url, endpointConfiguration))
            {
                var endpoints = client.GetEndpoints(null);
                return SelectEndpoint(url, endpoints, useSecurity);
            }
        }

```
and then 

 ```
private static Uri GetDiscoveryUrl(string discoveryUrl)
        {
            // needs to add the '/discovery' back onto non-UA TCP URLs.
            if (discoveryUrl.StartsWith(Utils.UriSchemeHttp))
            {
                if (!discoveryUrl.EndsWith("/discovery"))
                {
                    discoveryUrl += "/discovery";
                }
            }
            // parse the selected URL.
            return new Uri(discoveryUrl); //Problem here
        }

```
the call to  new Uri(discoveryUrl) always returns adding a / at the end and on "some" opc servers it fails . ( Not all  but in some it passes)

In case of node.js OPC the call is successful since / is not added explicitly . if we add / as part of ura there also call fails .

kevinherron, 2021-03-04:
Yes but if this fails when connecting to a Milo-based server I’d like to know where. It should at least get past the Hello message that contains the endpoint URL with the forward slash because there is logic to accommodate it there.

jsebes1, 2021-03-04:
Sorry , this happens probably not based on milo based server . it is opc client to a PLC communication. 
So the opc .net stack is used for opc client.

here Hello Passes and ACK happens , what really fails is create session
is it a wrong forum since I talked about C# opc client , but probably the bug is in C#  based opc stack used by client.

kevinherron, 2021-03-04:
Okay, that's what I thought might be happening. You should probably open a ticket over in the C# .NET stack repository.