eclipse-milo/milo #224: How to set Value Rank for any input/output argument

ravishsunny, 2018-02-07:
I needed to have an array of integers as input and output arguments. I understand that for that we need to set Value Rank as 1. But I am not able to find parameter for the same.

kevinherron, 2018-02-07:
Do you have any code you can show?

ravishsunny, 2018-02-07:
public class ConcatinateMethod {

    private final Logger logger = LoggerFactory.getLogger(getClass());

    @UaMethod
    public void invoke(
        InvocationContext context,

        @UaInputArgument(
            name = "x",
            description = "An integer value.")
            Integer x,
			
		@UaInputArgument(
            name = "y",
            description = "A double value.")
            Double y,
			
		@UaInputArgument(
            name = "z",
            description = "A string value.")
            String z,
			
		@UaInputArgument(
            name = "b",
            description = "A string value.")
            Boolean b,
			
		@UaInputArgument(
            name = "argList",
            description = "A string value.",
			rank  = "1")  // need your help here
            Integer argList,

kevinherron, 2018-02-07:
Change one of the arguments to be an array, the ValueRank will be changed for you.

ravishsunny, 2018-02-07:
You mean something like this?

@UaInputArgument(
            name = "argList",
            description = "Integer array")
            Integer argList[],

kevinherron, 2018-02-07:
Close... you need to use legal Java syntax...

```java
	@UaInputArgument(
        name = "argList",
        description = "A integer array.",
		rank  = "1")  // need your help here
        Integer[] argList,
```

ravishsunny, 2018-02-07:
This input should work, right? 

{"x":"78","y":"22.0","z":"Concatinating all","b":"false","argList":"1,2,3,4"}

kevinherron, 2018-02-07:
It should, assuming the client you're using correctly turns it from that JSON format to an actual OPC UA method call with the right data types.

ravishsunny, 2018-02-07:
value rank is set fine but the method call is failing with BadInternalError status when I am having an array as input argument. With others it is working fine. Any idea what could be the reason. I am using Unified Automation for client

kevinherron, 2018-02-07:
Is there an error in the logs? Can you post the full code sample?

I have created a test method with an array argument and it works fine calling from UaExpert.

ravishsunny, 2018-02-08:
UaStatus           status;
	OpcUa_UInt32       i = 0;
	CallIn             callRequest;
	CallOut            callResult;
	UaVariant          vTemp;
	ServiceSettings    serviceSettings;

	callRequest.objectId = *nodeId;
	callRequest.methodId = methodNodeId;
	UaArguments inputArguments;
	UaArguments outputArguments;

	m_uaSessionPtr->getMethodArguments(serviceSettings, callRequest.methodId, inputArguments, outputArguments);
	if (inputArguments.length() > 0)
	{
		// Create input arguments value array
		callRequest.inputArguments.create(inputArguments.length());
		for (i = 0; i < inputArguments.length(); i++)
		{
			UaString argumentName(inputArguments[i].Name);
			if (inputArgumentsMap.find(data_t(argumentName.toUtf8())) != inputArgumentsMap.end())
			{
				std::string argumentValue = MapHelper::getString(inputArgumentsMap, argumentName.toUtf8());
				UaVariant tempValue(argumentValue.c_str());

				if (inputArguments[i].ValueRank == 1)
				{
					tempValue.changeType((OpcUa_BuiltInType)inputArguments[i].DataType.Identifier.Numeric, OpcUa_True);
				}
				else
					tempValue.changeType((OpcUa_BuiltInType)inputArguments[i].DataType.Identifier.Numeric, OpcUa_False);

				tempValue.copyTo(&callRequest.inputArguments[i]);
			}
		}
	}

	status = m_uaSessionPtr->call(
		serviceSettings,    // Use default settings
		callRequest,        // In parameters and settings for the method call
		callResult);        // Out parameters and results returned from the method call

	
.
.
.
.

	if (status.isBad())
	{
		m_loggerPtr->debug("UaSession:: Method call failed [ret=%s]", status.toString().toUtf8());
		mapDeviceOperationResult.insert(data_t("message"), data_t(status.toString().toUtf8()));
	}
	else
	{
		m_loggerPtr->debug("UaSession:: Method call succeeded.");

		if (outputArguments.length() != callResult.outputArguments.length())
		{
			m_loggerPtr->info("Number of output arguments does not match OutputArguments property.");
			mapDeviceOperationResult.insert(data_t("message"), data_t("Number of output arguments does not match OutputArguments property."));
		}
		else
		{
			for (i = 0; i< callResult.outputArguments.length(); i++)
			{
				UaString argumentName(outputArguments[i].Name);
				vTemp = callResult.outputArguments[i];
				if (vTemp.type() == OpcUaType_ExtensionObject)
				{
					data_t retVal = m_ptransport->getUtils()->populateExtensionObjects(vTemp);
					resultMap.insert(data_t(argumentName.toUtf8()), std::move(retVal));
				}
				else
				{
					resultMap.insert(data_t(argumentName.toUtf8()), data_t(vTemp.toString().toUtf8()));
				}
			}
			mapDeviceOperationResult.insert(data_t("result"), data_t(std::move(resultMap)));
		}
	}

kevinherron, 2018-02-12:
Sorry, I meant the Milo code. I can't help you with C++ code written against the Unified Automation SDK.

I verified array arguments work correctly in Milo. Try calling the method with UaExpert, and if that works then you'll have to look closely at your client code.

closed 2018-02-12 by kevinherron