eclipse-milo/milo #225: Implement IssuedIdentityToken and corresponding validator

mbruns42, 2018-02-09:
The IssuedIdentityToken is specified in the OPC UA specification Part 4, 7.36.5. It should be added to `/opc-ua-sdk/sdk-server/src/main/java/org/eclipse/milo/opcua/sdk/server/api/config/OpcUaServerConfig.java`

A correpsonding validator should be added as a subclass of AbstractIdentityValidator.

mbruns42, 2018-02-09:
Why is there a special UserTokenType Kerberos @kevinherron ?

kevinherron, 2018-02-09:
I think it's a remnant of an older version of the OPC UA spec. These structures are all auto-generated.

mbruns42, 2018-02-09:
Yeah, I checked an older version, there used to be just Kerberos and WS, then they deprecated WS in 1.03 and now it's Kerberos, JWT and OAuth2. I'm struggeling a bit to get the ids right now. How did you come up with the Identifiers class?

kevinherron, 2018-02-09:
Identifiers is auto-generated from a CSV file that has all the NodeIds for OPC UA 1.03. You shouldn't need to add any or create any new structures to implement this.

kevinherron, 2018-05-25:
I'm going to close this for now. Validating an issued identity seems very application-specific and extending `AbstractIdentityValidator` is an easy enough job (one method to implement) that the SDK can't really add any more value.

If implement issued identity support at some point and find this conclusion to be incorrect let me know.

closed 2018-05-25 by kevinherron