eclipse-milo/milo #226: Eclipse Milo™ - an open source implementation of OPC UA (IEC 62541)

samber86, 2018-02-11:
Hello to all,

There are soo much errors in this project Eclipse milo.. it is about 4400 error.

can you please tell wheather i have to run this project on which version of eclipse

or what exactly is the problem

with best regards

Sam

ctron, 2018-02-12:
Hi @samber86 

it would be helpful if you provide a bit of information about what is wrong. Milo works fine for me and fine for many others.

A proper bug report would help to understand what exactly doesn't work for you. What the specific error is or what should be different in your opinion. "There are soo much errors…" doesn't help much to understand your problem and people might just ignore your report after all.

So I would kindle ask you to provide a bit of information if you expect that other people will invest their time in fixing your issue.

samber86, 2018-02-12:
[image]

see all the errors

samber86, 2018-02-12:
Hello ctron,

One for question what is milo exactly, is it a eclipse version or just a name

And can you tell me please which version of eclipse i can use to open this project

with best regards

Saurabh

samber.research@gmail.com

samber86, 2018-02-12:
These are details of error that are coming in every code, please tell me what should i will do

[image]

ctron, 2018-02-12:
Hi @samber86 

To me it looks like as if you didn't correctly import the projects into Eclipse. Use "Import …" -> "Maven / Existing Maven Projects" and point the wizard to the root directory of the checked out Milo repository.

I strongly recommend to start with a fresh workspace and git clone as it might be that your workspace already contains configurations and settings which prevent the import process from working correctly.

samber86, 2018-02-12:
I have to export the project..

but i have open the project like this.... open--java project and then uncheck the directory box and then open the project...

can you please explain me the steps what exactly i have to do..actually i am new in this field.

thanks in advance

best regards

Sam

ctron, 2018-02-12:
No, please follow the steps I mentioned above:

> Use "Import …" -> "Maven / Existing Maven Projects" and point the wizard to the root directory of the checked out Milo repository.
>
> I strongly recommend to start with a fresh workspace and git clone [...]

kevinherron, 2018-02-12:
@ctron thanks for taking the time to help him out.

@samber86 future questions like this are best handled on the mailing list: https://accounts.eclipse.org/mailing-list/milo-dev

samber86, 2018-02-12:
Thanks ctron, i have followed your step and now my errors are reduced to just 40...but the errors are not inside the OPC UA Stack and OPC UA SDK

[image]

Can you please tell me in this code what is bsd-parser-core and bsd-parser-json
And do i need these two in OPC UA client development

One more thing you can see in this screen short under milo folder there is sdk-client, sdk-server and stack client, stack-server

but what is outside the milo folder.. there are also the same folder sdk-client , sdk-server and stack

please explain me

thanks a lot for the help

best regards

Saurabh Samber

ctron, 2018-02-13:
Good point: 

> @samber86 future questions like this are best handled on the mailing list: https://accounts.eclipse.org/mailing-list/milo-dev

samber86, 2018-02-13:
I dint able to find where i can post my question, i have already subscribe the milo-dev

can you please tell me where i can post the question

closed 2018-02-16 by kevinherron