eclipse-milo/milo #227: Browse Server Tree not complete.

zulu1968, 2018-02-16:
Hi,
I encounter issues browsing the tag tree from a OPC-UA server. It seems I do not get all available tags from the server. I check with a OPC-UA client and get more results as when I use the browse method supplied. I tried to describe my issue in detail in attached Pdf file.

Regards,
Harry

[IssueOPC-UA.pdf](https://github.com/eclipse/milo/files/1731761/IssueOPC-UA.pdf)

kevinherron, 2018-02-16:
I can't tell if the issue is in your client code or the server from the information provided, but since we use Milo in our product at work and I've got a Siemens PLC with embedded UA server sitting on my desk right now, and I'm able to browse all nodes without any issue, I suspect the issue is in your code somewhere...

Can you provide a complete/runnable example? Maybe a wireshark capture of the browse request/response?

zulu1968, 2018-02-16:
Hi Kevin,

I tried also with the supplied examples at org.eclipse.milo.examples.client but same result:

[cid:image001.png@01D3A753.61CE71C0]

So this rules out custom code.

Although when using the nodeId ("SiteCom") direct to browse for tags it works normally. Also reading and writing values to these tags it all functions as expected.
The only issue is all tags enclosed in double quotes ("SiteCom"."MaterialNumber_1") don't show up when browsing the tree from the root folder (Identifiers.RootFolder)

NodeId{ns=3, id="SiteCom"}
NodeId{ns=3, id="SiteCom"."MaterialNumberBag"}
NodeId{ns=3, id="SiteCom"."MaterialNumber_1"}
NodeId{ns=3, id="SiteCom"."MaterialNumber_2"}
NodeId{ns=3, id="SiteCom"."MaterialNumber_3"}
NodeId{ns=3, id="SiteCom"."MaterialNumber_4"}
NodeId{ns=3, id="SiteCom"."MaterialNumber_5"}
NodeId{ns=3, id="SiteCom"."NumOfTriggers"}
NodeId{ns=3, id="SiteCom"."NumOfReads"}
NodeId{ns=3, id="SiteCom"."NumOfNoReads"}
NodeId{ns=3, id="SiteCom"."NumOfMatches"}
NodeId{ns=3, id="SiteCom"."NumOfMismatches"}
NodeId{ns=3, id="SiteCom"."Reading"}

Regards,
Harry

Essity Internal

This message may contain confidential, proprietary or legally
privileged information and is intended for the addressee's use
only. Any usage, disclosure, distribution, print or copying of
any part of this message is prohibited unless you are the
intended recipient. If you receive this message in error,
please delete it from any computer and notify the sender. If
you suspect that this message may have been altered, please
notify the sender. Essity has taken every reasonable precaution
to ensure that this e-mail and any attachments to this e-mail
has been scanned for viruses. However, Essity does not accept
liability for any damage caused by software viruses brought to
you by this mail.

zulu1968, 2018-02-16:
[image]

Forgot the picture ;)

kevinherron, 2018-02-16:
Modify BrowseExample so the BrowseDescription it builds looks like this:

```java
        BrowseDescription browse = new BrowseDescription(
            browseRoot,
            BrowseDirection.Forward,
            Identifiers.HierarchicalReferences,
            true,
            uint(NodeClass.Object.getValue() | NodeClass.Variable.getValue()),
            uint(BrowseResultMask.All.getValue())
        );
```

The difference being `Identifiers.HierarchicalReferences` vs `Identifiers.References`.

zulu1968, 2018-02-16:
Same result..

Essity Internal

________________________________

This message may contain confidential, proprietary or legally
privileged information and is intended for the addressee's use
only. Any usage, disclosure, distribution, print or copying of
any part of this message is prohibited unless you are the
intended recipient. If you receive this message in error,
please delete it from any computer and notify the sender. If
you suspect that this message may have been altered, please
notify the sender. Essity has taken every reasonable precaution
to ensure that this e-mail and any attachments to this e-mail
has been scanned for viruses. However, Essity does not accept
liability for any damage caused by software viruses brought to
you by this mail.

kevinherron, 2018-02-16:
I'll need a Wireshark of the traffic between Milo and the PLC and whatever other client and the PLC. I can't see anything wrong with the code, you're just not getting the references in return for some reason.

kevinherron, 2018-02-16:
Here's a snippet of the browse against the S7-1500 I have on my desk: 

```
10:04:09.336 [main] INFO  o.e.m.examples.client.BrowseExample -    Node=DeviceSet NodeId=NodeId{ns=2, id=5001}
10:04:09.341 [main] INFO  o.e.m.examples.client.BrowseExample -      Node=PLC_2 NodeId=NodeId{ns=3, id=PLC}
10:04:09.347 [main] INFO  o.e.m.examples.client.BrowseExample -        Node=Counters NodeId=NodeId{ns=3, id=Counters}
10:04:09.355 [main] INFO  o.e.m.examples.client.BrowseExample -        Node=DataBlocksGlobal NodeId=NodeId{ns=3, id=DataBlocksGlobal}
10:04:09.361 [main] INFO  o.e.m.examples.client.BrowseExample -          Node=Data_block_1 NodeId=NodeId{ns=3, id="Data_block_1"}
10:04:09.368 [main] INFO  o.e.m.examples.client.BrowseExample -            Node=IEC_COUNTER NodeId=NodeId{ns=3, id="Data_block_1"."IEC_COUNTER"}
10:04:09.374 [main] INFO  o.e.m.examples.client.BrowseExample -              Node=CU NodeId=NodeId{ns=3, id="Data_block_1"."IEC_COUNTER".CU}
10:04:09.380 [main] INFO  o.e.m.examples.client.BrowseExample -              Node=CD NodeId=NodeId{ns=3, id="Data_block_1"."IEC_COUNTER".CD}
10:04:09.386 [main] INFO  o.e.m.examples.client.BrowseExample -              Node=R NodeId=NodeId{ns=3, id="Data_block_1"."IEC_COUNTER".R}
10:04:09.393 [main] INFO  o.e.m.examples.client.BrowseExample -              Node=LD NodeId=NodeId{ns=3, id="Data_block_1"."IEC_COUNTER".LD}
10:04:09.403 [main] INFO  o.e.m.examples.client.BrowseExample -              Node=QU NodeId=NodeId{ns=3, id="Data_block_1"."IEC_COUNTER".QU}
10:04:09.409 [main] INFO  o.e.m.examples.client.BrowseExample -              Node=QD NodeId=NodeId{ns=3, id="Data_block_1"."IEC_COUNTER".QD}
10:04:09.416 [main] INFO  o.e.m.examples.client.BrowseExample -              Node=PV NodeId=NodeId{ns=3, id="Data_block_1"."IEC_COUNTER".PV}
10:04:09.422 [main] INFO  o.e.m.examples.client.BrowseExample -              Node=CV NodeId=NodeId{ns=3, id="Data_block_1"."IEC_COUNTER".CV}
10:04:09.428 [main] INFO  o.e.m.examples.client.BrowseExample -            Node=bool NodeId=NodeId{ns=3, id="Data_block_1"."bool"}
10:04:09.435 [main] INFO  o.e.m.examples.client.BrowseExample -            Node=IEC_TIMER NodeId=NodeId{ns=3, id="Data_block_1"."IEC_TIMER"}
10:04:09.442 [main] INFO  o.e.m.examples.client.BrowseExample -              Node=PT NodeId=NodeId{ns=3, id="Data_block_1"."IEC_TIMER".PT}
10:04:09.449 [main] INFO  o.e.m.examples.client.BrowseExample -              Node=ET NodeId=NodeId{ns=3, id="Data_block_1"."IEC_TIMER".ET}
10:04:09.456 [main] INFO  o.e.m.examples.client.BrowseExample -              Node=IN NodeId=NodeId{ns=3, id="Data_block_1"."IEC_TIMER".IN}
10:04:09.464 [main] INFO  o.e.m.examples.client.BrowseExample -              Node=Q NodeId=NodeId{ns=3, id="Data_block_1"."IEC_TIMER".Q}
10:04:09.471 [main] INFO  o.e.m.examples.client.BrowseExample -            Node=int NodeId=NodeId{ns=3, id="Data_block_1"."int"}
10:04:09.477 [main] INFO  o.e.m.examples.client.BrowseExample -            Node=lint NodeId=NodeId{ns=3, id="Data_block_1"."lint"}
10:04:09.484 [main] INFO  o.e.m.examples.client.BrowseExample -            Node=real NodeId=NodeId{ns=3, id="Data_block_1"."real"}
10:04:09.490 [main] INFO  o.e.m.examples.client.BrowseExample -            Node=instance_FooBarBaz NodeId=NodeId{ns=3, id="Data_block_1"."instance_FooBarBaz"}
10:04:09.497 [main] INFO  o.e.m.examples.client.BrowseExample -              Node=foo NodeId=NodeId{ns=3, id="Data_block_1"."instance_FooBarBaz"."foo"}
10:04:09.504 [main] INFO  o.e.m.examples.client.BrowseExample -              Node=bar NodeId=NodeId{ns=3, id="Data_block_1"."instance_FooBarBaz"."bar"}
10:04:09.511 [main] INFO  o.e.m.examples.client.BrowseExample -              Node=baz NodeId=NodeId{ns=3, id="Data_block_1"."instance_FooBarBaz"."baz"}
10:04:09.519 [main] INFO  o.e.m.examples.client.BrowseExample -              Node=array NodeId=NodeId{ns=3, id="Data_block_1"."instance_FooBarBaz"."array"}
10:04:09.525 [main] INFO  o.e.m.examples.client.BrowseExample -                Node=0 NodeId=NodeId{ns=3, id="Data_block_1"."instance_FooBarBaz"."array"[0]}
10:04:09.531 [main] INFO  o.e.m.examples.client.BrowseExample -                Node=1 NodeId=NodeId{ns=3, id="Data_block_1"."instance_FooBarBaz"."array"[1]}
10:04:09.538 [main] INFO  o.e.m.examples.client.BrowseExample -                Node=2 NodeId=NodeId{ns=3, id="Data_block_1"."instance_FooBarBaz"."array"[2]}
10:04:09.544 [main] INFO  o.e.m.examples.client.BrowseExample -                Node=3 NodeId=NodeId{ns=3, id="Data_block_1"."instance_FooBarBaz"."array"[3]}
```

Are you sure you're connected to a server that actually has these nodes? What's the firmware version?

zulu1968, 2018-02-16:
PLC Type: Siemens	6ES7 511-1AK01-0AB0 
Software version: V02.05.00

I also connect to the PLC with a OPC-UA client and here I see the tags. This indicates the tags are available, also I'm able to write to them using Milo so the connection is ok. And using the same credentials on both the Client and Milo.

At the moment I can not create a WhireShark dump, as the current connection is encrypted, I don't think this is usefull. I'm only able to change this to no encription after the weekend.

kevinherron, 2018-02-16:
You'll have to compare Wireshark captures to see what the difference is. Maybe one has a username and one doesn't?

zulu1968, 2018-02-19:
[wireshark.zip](https://github.com/eclipse/milo/files/1737470/wireshark.zip)

Hi, I captured from both Prosys and Milo the request to this particular node. And you can see the response is different. Or maybe I captured the wrong thing 😉

Somehow Prosys does a BrowseNextRequest call where we get the desired info we miss in Milo.

If you need additional captures please let me know.

kevinherron, 2018-02-19:
Ok, that explains it. Milo SDK leaves it up to the application to call BrowseNext if needed. I should probably augment the example to show this.

zulu1968, 2018-02-19:
But that does not explain, if you used the same BrowseExample, you read direct all the nodes. And I should do this BrowseNext call on the same type of equipment. Or is this BrowseNext call an setup item?

kevinherron, 2018-02-19:
My test PLC must have less nodes, so the server did not respond with a continuation point to call BrowseNext against.

Modify the browse example like this and tell me if it works:

```java
    private void browseNode(String indent, OpcUaClient client, NodeId browseRoot) {
        BrowseDescription browse = new BrowseDescription(
            browseRoot,
            BrowseDirection.Forward,
            Identifiers.References,
            true,
            uint(NodeClass.Object.getValue() | NodeClass.Variable.getValue()),
            uint(BrowseResultMask.All.getValue())
        );

        try {
            BrowseResult browseResult = client.browse(browse).get();

            List<ReferenceDescription> references = new CopyOnWriteArrayList<>();
            references.addAll(toList(browseResult.getReferences()));

            ByteString continuationPoint = browseResult.getContinuationPoint();

            while (continuationPoint != null && continuationPoint.isNotNull()) {
                BrowseResult nextResult = client.browseNext(true, continuationPoint).get();

                references.addAll(toList(nextResult.getReferences()));

                continuationPoint = nextResult.getContinuationPoint();
            }

            for (ReferenceDescription rd : references) {
                logger.info("{} Node={}", indent, rd.getBrowseName().getName());

                // recursively browse to children
                rd.getNodeId().local().ifPresent(nodeId -> browseNode(indent + "  ", client, nodeId));
            }
        } catch (InterruptedException | ExecutionException e) {
            logger.error("Browsing nodeId={} failed: {}", browseRoot, e.getMessage(), e);
        }
    }
```

zulu1968, 2018-02-19:
[Milo-2.zip](https://github.com/eclipse/milo/files/1737608/Milo-2.zip)
Nope,

Although we see now indeed a subsequent call, but still no information :(

kevinherron, 2018-02-19:
Okay, one more try:

```java
    private void browseNode(String indent, OpcUaClient client, NodeId browseRoot) {
        BrowseDescription browse = new BrowseDescription(
            browseRoot,
            BrowseDirection.Forward,
            Identifiers.HierarchicalReferences,
            true,
            uint(0),
            uint(BrowseResultMask.All.getValue())
        );

        try {
            BrowseResult browseResult = client.browse(browse).get();

            List<ReferenceDescription> references = new CopyOnWriteArrayList<>();
            references.addAll(toList(browseResult.getReferences()));

            ByteString continuationPoint = browseResult.getContinuationPoint();

            while (continuationPoint != null && continuationPoint.isNotNull()) {
                BrowseResult nextResult = client.browseNext(true, continuationPoint).get();

                references.addAll(toList(nextResult.getReferences()));

                continuationPoint = nextResult.getContinuationPoint();
            }

            for (ReferenceDescription rd : references) {
                logger.info("{} Node={}", indent, rd.getBrowseName().getName());

                // recursively browse to children
                rd.getNodeId().local().ifPresent(nodeId -> browseNode(indent + "  ", client, nodeId));
            }
        } catch (InterruptedException | ExecutionException e) {
            logger.error("Browsing nodeId={} failed: {}", browseRoot, e.getMessage(), e);
        }
    }
```

This should produce a BrowseRequest identical to the Prosys request.

zulu1968, 2018-02-19:
[Milo-3.zip](https://github.com/eclipse/milo/files/1737671/Milo-3.zip)

Also not returning the nodes..

kevinherron, 2018-02-19:
> Also not returning the nodes..

Yes, sorry, please see the edit. The releaseContinuationPoints parameter must be false.

zulu1968, 2018-02-19:
[Milo-4.zip](https://github.com/eclipse/milo/files/1737689/Milo-4.zip)
 Yes, this did the trick :+1: 
Thanks!

closed 2018-02-20 by kevinherron