eclipse-milo/milo #230: Building the project

sergejzr, 2018-02-27:
Hello, dear contributors!

I cloned the project, executed "mvn clean install" as suggested by the readme, however it could not build. Can you tell me what went wrong (see maven ouptut below)?

> [INFO] Scanning for projects...
[ERROR] [ERROR] Some problems were encountered while processing the POMs:
[FATAL] Non-resolvable parent POM for org.eclipse.milo:standalone-examples:[unknown-version]: Could not find artifact org.eclipse.milo:milo-examples:pom:0.2.1-SNAPSHOT and 'parent.relativePath' points at wrong local POM @ line 6, column 13
 @ 
[ERROR] The build could not read 1 project -> [Help 1]
[ERROR]   
[ERROR]   The project org.eclipse.milo:standalone-examples:[unknown-version] (/home/zerr/soft/milo/milo/milo-examples/standalone-examples/pom.xml) has 1 error
[ERROR]     Non-resolvable parent POM for org.eclipse.milo:standalone-examples:[unknown-version]: Could not find artifact org.eclipse.milo:milo-examples:pom:0.2.1-SNAPSHOT and 'parent.relativePath' points at wrong local POM @ line 6, column 13 -> [Help 2]
[ERROR] 
[ERROR] To see the full stack trace of the errors, re-run Maven with the -e switch.
[ERROR] Re-run Maven using the -X switch to enable full debug logging.
[ERROR] 
[ERROR] For more information about the errors and possible solutions, please read the following articles:
[ERROR] [Help 1] http://cwiki.apache.org/confluence/display/MAVEN/ProjectBuildingException
[ERROR] [Help 2] http://cwiki.apache.org/confluence/display/MAVEN/UnresolvableModelException

kevinherron, 2018-02-27:
Thank you, I think I just pushed a commit that will fix this.

sergejzr, 2018-02-27:
Thanks, Kevin! 
Unfortunately in the file milo-examples/standalone-examples/pom.xml are 3 more  <version>0.2.1-SNAPSHOT</version> that still give error. I changed to  0.2.2 (after looking at your commit) and now it works for me locally. You may want to introduce those changes to the repo as well.
Best,
Sergej

kevinherron, 2018-02-27:
Bleh, careless of me. Thanks.

closed 2018-02-27 by kevinherron