eclipse-milo/milo #238: NPE on reading nested struct with built-in data type

bdorninger, 2018-03-14:
Setup: Unified Automation Demo Server, linked custom model (NS=5), Milo 0.2.1 on JDK 1.8.121
server, the model export file (unified automation modeler) plus the client code can be found at this link: https://drive.google.com/open?id=1f00HVvx7rlquKuFvMeuBKEQDA0psAyh5

We have a struct containing an (optional) member being of a built-in datatype.
The struct looks like this:
PointStructType {
Double x;
Double y;
Range rangeX;
Range rangeY;
}
 
Although Milo can decode a Range object when it is read directly, Milo fails to decode when the range is embedded into another (extension) object. See stacktrace
Reason is, that in the AbstractCodecs READER/WRITER maps, there are only some builtin/core types, but Range is missing.
From the design, it’s clear that users are not encouraged to modify these maps :-)
 
 
java.lang.NullPointerException: null
       at org.eclipse.milo.opcua.binaryschema.AbstractCodec.decode(AbstractCodec.java:197)
       at org.eclipse.milo.opcua.stack.core.types.OpcUaDataTypeEncoding.decodeFromByteString(OpcUaDataTypeEncoding.java:126)
       at org.eclipse.milo.opcua.stack.core.types.builtin.ExtensionObject.decode(ExtensionObject.java:85)
       at org.eclipse.milo.opcua.stack.core.types.builtin.ExtensionObject.decode(ExtensionObject.java:69)
       at at.scch.opcua.milo.client.sample.ReadCustomDataTypeExample.rwDemoStruct(ReadCustomDataTypeExample.java:101)
       at at.scch.opcua.milo.client.sample.ReadCustomDataTypeExample.run(ReadCustomDataTypeExample.java:69)
       at at.scch.opcua.milo.client.sample.ClientExampleRunner.run(ClientExampleRunner.java:132)
       at at.scch.opcua.milo.client.sample.ReadCustomDataTypeExample.main(ReadCustomDataTypeExample.java:44)

kevinherron, 2018-03-19:
Which node is it that has this embedded Range?

Also - the server complains that it cannot find the file `uanodesetimport.xml` on starting - expected?

kevinherron, 2018-03-19:
Thanks for the server and code. I've identified and fixed the issue. I will push the fix once I've added a test case for this.

kevinherron, 2018-03-19:
@bdorninger if you could check out that branch and verify (or keep testing other structures) that would be great.

bdorninger, 2018-03-22:
H Kevin

Just tried to build the checked out gh-238 branch, but checkstyle reports some errors in bsd-parser-core, failing the build (Lines 197 and 278, Operator Placement)
Corrected these, then build went through fine.
Regards,
Bernhard

bdorninger, 2018-03-22:
Thx! Looks good now. 
Will test it asap with a custom Decoder (relying on Milo's built-in decoder for the nested type)

bdorninger, 2018-03-22:
no problem, since i did not pull it via git, just downloaded the branch as zip

closed 2018-04-20 by kevinherron