eclipse-milo/milo #239: Custom object and type instance with attribute delegate

rich0rd, 2018-03-19:
Hy,

I was playing with the server example of milo, more precisely with the addCustomObjectTypeAndInstance” method. (https://github.com/eclipse/milo/blob/c77d06a4d7ccfe408c3e931e15700f53c42a8f8d/milo-examples/server-examples/src/main/java/org/eclipse/milo/examples/server/ExampleNamespace.java#L540)

In my example my object nodes should have user access level AccessLevel.CurrentRead, AccessLevel.HistoryRead and an attribute delegate for reading requests.  
In the milo server example, the object node is created comfortable by the node factory.
But neither the user access level nor the attribute delegate is set/copied in the created object node.
I’m new to milo and could not access if the node factory should do this.
But who can I access the created child nodes of my object node to set these attributes?

```java
        // Use NodeFactory to create instance of MyObjectType called "MyObject".
        // NodeFactory takes care of recursively instantiating MyObject member nodes
        // as well as adding all nodes to the address space.
        UaObjectNode myObject = nodeFactory.createObject(
            new NodeId(namespaceIndex, "HelloWorld/MyObject"),
            new QualifiedName(namespaceIndex, "MyObject"),
            LocalizedText.english("MyObject"),
            objectTypeNode.getNodeId()
      );

      //how to create AccessLevel.CurrentRead, AccessLevel.HistoryRead and an attribute delegate 
      //for "foo" and "bar" node???
```
Many thanks,

Dalaitama, 2018-03-19:
Hallo 

IMHO, i’m not 100% sure but an attribute delegate doesn’t belong to an Instance Definition, it belongs to the ObjectNode which will be instantiated later. I didn’t find out which member are applied on the ObjectNode created on cascade from the nodeFactory, but an AttributeDelegate  if set on the InstanceDefinition, will not work on the instance. You have to set it after the object creation, on the real instance.

I assume this is not a Bug (may be I am wrong) but to me could make sense, since if you create 3 instances of a NodeType with the Node Factory (and its components, aka InstanceDefinitions)  , then I think you should set there  attribute delegate. Hence it will not make sense to have in the same server 3 Attribute Delegate which potentially access the same data. 

Make sense ? @kevinherron , i am in a wrong way?

Thanks Lorenzo

kevinherron, 2018-03-19:
Yes, that's correct, you would not put a delegate onto the definition for
your custom type. It belongs on an instance.

Further, AccessLevel is an Attribute that belongs to a VariableNode, not
and ObjectNode, which is why you aren't able to configure it.

rich0rd, 2018-03-21:
Many thanks for the quick response.
Of course the AccessLevel belongs to the VaribleNode but even if I set the AccessLevel on "foo" and "bar", the node factory does not assign the property to the object node.
See:
https://github.com/eclipse/milo/blob/c77d06a4d7ccfe408c3e931e15700f53c42a8f8d/opc-ua-sdk/sdk-server/src/main/java/org/eclipse/milo/opcua/sdk/server/nodes/NodeFactory.java#L196

kevinherron, 2018-03-21:
Sorry, I misunderstood your issue originally.

kevinherron, 2018-03-21:
Ok, it's a little more complicated. `VariableTypeNode`s don't *have* `AccessLevel` and `UserAccessLevel` attributes, which is why they aren't "inherited" or assigned when building an instance.

The "fix" for this is going to require some work I'm planning on the 0.3 branch, which is to allow some kind of interceptor/visitor to be installed on the factory that gets called/notified at various stages of construction of a complex node hierarchy. In this case, the way it might work is you'd get a call that provided you the parent node and the newly instantiated child node and you'd have an opportunity to do anything you wish with that information, such as set an `AccessLevel` on the node, or instantiate some type of object(s) specific to your application that the node represents, etc...

PR #318 (2018-10-16): InstanceDeclarationHierarchy-based NodeFactory
NodeFactory that instantiates nodes based on an InstanceDeclarationHierarchy as described in OPC UA Part 3, Sections 6.3 and 6.4.

fixes #239, #248
commit 66eed8a8 (2018-10-16): InstanceDeclarationHierarchy-based NodeFactory (#318)

NodeFactory that instantiates nodes based on an InstanceDeclarationHierarchy as described in OPC UA Part 3, Sections 6.3 and 6.4.

fixes #239, #248
PR #334 (2018-10-16): InstanceDeclarationHierarchy-based NodeFactory
NodeFactory that instantiates nodes based on an InstanceDeclarationHierarchy as described in OPC UA Part 3, Sections 6.3 and 6.4.

fixes #239, #248
commit 8247a824 (2018-10-16): InstanceDeclarationHierarchy-based NodeFactory (#334)

NodeFactory that instantiates nodes based on an InstanceDeclarationHierarchy as described in OPC UA Part 3, Sections 6.3 and 6.4.

fixes #239, #248