eclipse-milo/milo #241: Pass request timeout from the configuration to the client's session operations

miroch41, 2018-03-20:
Some methods in _org.eclipse.milo.opcua.sdk.client.session.states.AbstractSessionState_ (e.g. activateSession, createSession, ...) do not use request timeout specified in the client's configuration, but, instead, they use a hard-coded timeout of _REQUEST_TIMEOUT_=16 seconds. In my case, 16 seconds is too much. Although it is a good idea to use some reasonable request timeout if the configuration specifies none, it should be somehow configurable (that is, if for some reason the "common" request timeout from configuration is not suitable for use in session operations, then an additional configuration shall be provided for this purpose). Anyway, it should not be hard-coded. 

Note that using timed-get in the CompletableFuture _OpcUaClient.connect()_ via get(&lt;some timeout&gt;) is no workaround for this problem, since after such get-operation times-out:

- Client's trial to connect to the server is not necessarily (immediately) interrupted
- Timed-get leaves the client instance in an undefined state, not suitable for further use and problematical for cleanup.

kevinherron, 2018-03-20:
> that is, if for some reason the "common" request timeout from configuration is not suitable for use in session operations, then an additional configuration shall be provided for this purpose

Yes, that's the problem.

> Anyway, it should not be hard-coded.

Agreed. I think a "Connect Timeout" property that also gets used during establishment of the TCP connection (which is also hardcoded right now) could be used for the session operations.

kevinherron, 2018-03-23:
Split this out into #254 and #255

closed 2018-03-23 by kevinherron