eclipse-milo/milo #244: Regression with username/password authentication

ctron, 2018-03-20:
I think there is a regression in username password authentication from Milo 0.1.6 to 0.2.1 (maybe 0.2.0).

Having a simple username/password authentication without certificates like this:

~~~java
    public static void main(String[] args) throws Exception {

        final OpcUaServerConfig serverConfig = new OpcUaServerConfigBuilder() //
            .setSecurityPolicies(EnumSet.of(SecurityPolicy.None)) //
            .setIdentityValidator(new UsernameIdentityValidator(true, credentials -> {

                return credentials.getUsername().equals("foo") && credentials.getPassword().equals("bar");

            })) //
            .setCertificateManager(new DefaultCertificateManager()) //
            .setCertificateValidator(new CertificateValidator() {

                @Override
                public void verifyTrustChain(X509Certificate arg0, List<X509Certificate> arg1) throws UaException {
                }

                @Override
                public void validate(X509Certificate arg0) throws UaException {
                }
            }) //
            .setUserTokenPolicies(asList(USER_TOKEN_POLICY_USERNAME, USER_TOKEN_POLICY_ANONYMOUS)) //
            .build();

        final OpcUaServer server = new OpcUaServer(serverConfig);
        server.startup().get();

        // client

        EndpointDescription[] endpoints = UaTcpStackClient.getEndpoints("opc.tcp://localhost:12685").get();
        for (EndpointDescription desc : endpoints) {
            System.out.println(desc);
        }

        Optional<EndpointDescription> ep = //
            stream(endpoints) //
                .filter(e -> e.getSecurityPolicyUri().equals(SecurityPolicy.None.getSecurityPolicyUri())) //
                .findFirst();

        final OpcUaClientConfig clientConfig = new OpcUaClientConfigBuilder() //
            .setIdentityProvider(new UsernameProvider("foo", "bar")) //
            // .setIdentityProvider(new AnonymousProvider()) //
            .setEndpoint(ep.get()) //
            .build();

        OpcUaClient client = new OpcUaClient(clientConfig);

        client //
            .connect().get();
~~~

was working in 0.1.6. But with 0.2.1 you get:

~~~
Exception in thread "main" java.util.concurrent.ExecutionException: java.lang.NullPointerException: certificateBytes cannot be null
	at java.util.concurrent.CompletableFuture.reportGet(CompletableFuture.java:357)
	at java.util.concurrent.CompletableFuture.get(CompletableFuture.java:1895)
	at Application.main(Application.java:75)
Caused by: java.lang.NullPointerException: certificateBytes cannot be null
	at com.google.common.base.Preconditions.checkNotNull(Preconditions.java:228)
	at org.eclipse.milo.opcua.stack.core.util.CertificateUtil.decodeCertificate(CertificateUtil.java:47)
	at org.eclipse.milo.opcua.sdk.client.api.identity.UsernameProvider.getIdentityToken(UsernameProvider.java:98)
	at org.eclipse.milo.opcua.sdk.client.session.states.AbstractSessionState.lambda$activateSession$7(AbstractSessionState.java:231)
	at java.util.concurrent.CompletableFuture.uniComposeStage(CompletableFuture.java:981)
	at java.util.concurrent.CompletableFuture.thenCompose(CompletableFuture.java:2124)
	at org.eclipse.milo.opcua.sdk.client.session.states.AbstractSessionState.activateSession(AbstractSessionState.java:220)
	at org.eclipse.milo.opcua.sdk.client.session.states.AbstractSessionState.lambda$activateSessionAsync$5(AbstractSessionState.java:207)
	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1149)
	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:624)
	at java.lang.Thread.run(Thread.java:748)
~~~

kevinherron, 2018-03-20:
I'll take a look. Probably an easy fix. Thank you.

kevinherron, 2018-03-20:
Okay, so I should still fix this so it doesn't NPE, but it seems to stem from an "invalid" server configuration in which the server has set up a user token policy that specifies encryption but is not providing a certificate in the endpoint.

kevinherron, 2018-03-20:
@ctron can you see if this commit improves the error you see?

It may also be worth exploring how to prevent the server from being misconfigured, but I'd like to take care of the NPE first.

ctron, 2018-03-21:
I will give it a try. But what I don't understand is, in the past I didn't use any encryption either. Using the same code, switching between 0.1.6 and 0.2.1 brings up this error.

ctron, 2018-03-21:
So with that commit the error simple changed to:

~~~
Exception in thread "main" java.util.concurrent.ExecutionException: UaException: status=Bad_ConfigurationError, message=UserTokenPolicy requires encryption but server did not provide a certificate in endpoint
	at java.util.concurrent.CompletableFuture.reportGet(CompletableFuture.java:357)
	at java.util.concurrent.CompletableFuture.get(CompletableFuture.java:1895)
	at Application.main(Application.java:80)
Caused by: UaException: status=Bad_ConfigurationError, message=UserTokenPolicy requires encryption but server did not provide a certificate in endpoint
	at org.eclipse.milo.opcua.sdk.client.api.identity.UsernameProvider.getIdentityToken(UsernameProvider.java:100)
	at org.eclipse.milo.opcua.sdk.client.session.states.AbstractSessionState.lambda$activateSession$7(AbstractSessionState.java:231)
	at java.util.concurrent.CompletableFuture.uniComposeStage(CompletableFuture.java:981)
	at java.util.concurrent.CompletableFuture.thenCompose(CompletableFuture.java:2124)
	at org.eclipse.milo.opcua.sdk.client.session.states.AbstractSessionState.activateSession(AbstractSessionState.java:220)
	at org.eclipse.milo.opcua.sdk.client.session.states.AbstractSessionState.lambda$activateSessionAsync$5(AbstractSessionState.java:207)
	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1149)
	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:624)
	at java.lang.Thread.run(Thread.java:748)
~~~

Now dug a bit trough the code and saw that around 4 months ago you have changed the security policy of `USER_TOKEN_POLICY_USERNAME` from `null` to `Basic256` (see 1f8de7cde0f2a57979d0292d7887a4700058a03c). Which would explain the changed behavior.

So I did change my code to use a custom `UserTokenPolicy` and it worked again.

However, setting multiple policies seems to be unsupported:
~~~java
.setUserTokenPolicies(asList(USER_TOKEN_POLICY_USERNAME, USER_TOKEN_POLICY_ANONYMOUS, myUserTokenPolicy))
~~~

did only result in the "first" one failing with the same error.

So I think the change you made in 7c9801644bbd79ce43f6d26384cfc0b716059e39 does improve the situation. I also think that is might have been a good idea to keep the definition of the "old" `USER_TOKEN_POLICY_USERNAME` the same over releases. But I guess that is too late.

However I would expect that I can add multiple policies for the same type (`username`), which all get evaluated. Of course I am not sure if OPC UA allows for that.

But thanks for looking into this!

kevinherron, 2018-03-21:
> I also think that is might have been a good idea to keep the definition of the "old" USER_TOKEN_POLICY_USERNAME the same over releases. But I guess that is too late.

The old token policy was insecure and could lead to a password being sent in plaintext. Ideally, even if your server will allow no security connections, if you're going to use username/password authentication you configure a certificate as well. I at least made the change between 0.1.x and 0.2.x (I think).

> However I would expect that I can add multiple policies for the same type (username), which all get evaluated. Of course I am not sure if OPC UA allows for that.

It's allowed, but I think the client logic that searches for an endpoint with the desired user token policy just uses the first one it finds.

ctron, 2018-03-21:
I agree that the change was in the right direction!

Yes, so maybe we should close this issue and create a new feature request issue, for the ability to user multiple `UserTokenPolicy`s.

kevinherron, 2018-03-21:
I'll close this when I merge the change, the NPE that happened was definitely an issue.

Letting the user (optionally) select the token policy should be pretty easy as well.

closed 2018-03-22 by kevinherron