eclipse-milo/milo #248: NodeFactory and ObjectTypes with Subtypes

rich0rd, 2018-03-21:
I have the following ObjectType definitions:
`SensorValue` (ObjectType) with variable “value” (double)
`SensorValueQuality` (ObjectType) with variable “qualityInformation” (int32)
`SensorValueQuality` is a sub type of `SensorValue`
 
If I create an instance of the type `SensorValueQuality` I would expect an Object with the two variables “value” and “qualityInformation”. 
But instances created by the `NodeFactory` only contains the variable “qualityInformation”, variable “value” is missing.
Isn't the `NodeFactory` responsible for creating the variables of the sub types?

Many thanks,

kevinherron, 2018-03-21:
I'll look into this. I agree that instantiating a subtype should result in members of its parent type being present as well.

kevinherron, 2018-03-22:
This is a bit tricky to fix/implement, and I may do it in the `dev/0.3` branch because modeling is one of the things I want to focus on in the 0.3 release.

kevinherron, 2018-04-26:
Note to self: OPC UA Part 3, section 6.3.3

...complicated is maybe an understatement.

kevinherron, 2018-09-09:
@rich0rd I don't know if you're still around or using this library, but if you wanted to provide feedback on #318 now would be the time to do it.