eclipse-milo/milo #257: No input/output-arguments for undefined methods

miroch41, 2018-03-26:
OPC-UA server browse-tree contains multiple method-nodes. Not all of them are currently implemented (actually, only few of them are). 

Those method-nodes, which are not implemented yet (take SetSubscriptionDurable as an example), contain subnodes for "InputArguments" and "OutputArguments", however, the respective argument types are not available within these subnodes. The fact, that the method implementation is not available yet, is understandable, but without argument types (which are rather a part of method definition than its functionality) common clients _crash_ prior to method call, as they do not expect that the argument types of the method are missing. I would rather expect that a client is informed that he actully cannot make use of the method (either in advance or via return info, e.g. as  status code _Bad_ServiceUnsupported_).

Current situation is caused by the fact that the method-nodes (and their subnodes) are created statically, but the definition of argument types is inferred from the method implementation using  _org.eclipse.milo.opcua.sdk.server.util.AnnotationBasedInvocationHandler_ when constructing the namespace: if there is no implementation at all (neither an empty one), argument types are not inferred.

kevinherron, 2018-03-26:
I think you're correct that there's a problem here, but the analysis is not right in this case. `AnnotationBasedInvocationHandler` simply isn't used for any of the unimplemented methods.

I think the problem has to do with `ExtensionObject`s being double-encoded. You can see how the SetSubscriptionDurable method is loaded in `UaMethodLoader:228` and its type nodes in `UaVariableLoader:1732` - this all seems fine. 

But I think the problem is we're putting an `ExtensionObject` into a `Variant` as we build the model rather than the decoded value (which *should* be fine), but when that Variant is later re-encoded when the node is read, the encoder doesn't recognize it's already an ExtensionObject and re-encodes it as one... or something like that.

In this screenshot from UaExpert you can see the SetDurableSubscription.InputArguments value is actually there, but it's an array of `ExtensionObject` rather than an array of `Argument`.

kevinherron, 2018-05-25:
I made a little progress on this in the `dev/0.3` branch. The wrong XML content was being sent inside the ExtensionObject.

After making the changes I was able to read the values using the .NET reference client so I think I'm headed in the right direction.

Unfortunately it still doesn't work quite right with UaExpert. I think it just doesn't like that I'm serving up an XML-encoded ExtensionObject when the transport is set up for binary encoding.

So, what's left then, is probably a related issue where the server should to respect the requested data encoding (which when unspecified in the service call defaults to the transport encoding).

closed 2018-05-31 by kevinherron