eclipse-milo/milo #260: Provide a way to override endpoint information

ctron, 2018-05-03:
Following up from the gitter conversion with @mbruns42 

In some scenarios the server reports back with a hostname which the client doesn't understand. Assume you have an OPC UA server running inside a docker container. Connecting to "host-1" and fetching endpoint information might return endpoints with the name "container-1", which the client cannot use. So the client has to override the provided information with the original hostname it used ("host-1") in order to connect.

Currently I do this in `camel-milo` like this: https://github.com/apache/camel/blob/master/components/camel-milo/src/main/java/org/apache/camel/component/milo/client/internal/SubscriptionManager.java#L576-L587

However I do think that there should be a simpler way to achieve the same result. Some out-of-the-box support for overriding various parts of the endpoint information.

kevinherron, 2018-05-03:
There is already configuration for this: https://github.com/eclipse/milo/blob/b4237b40d81de977ddc1d79f15590d172d9fcbeb/opc-ua-sdk/sdk-server/src/main/java/org/eclipse/milo/opcua/sdk/server/api/config/OpcUaServerConfig.java#L69-L77

The effective list of `EndpointDescription` is the product of bind addresses and endpoint addresses (and security profiles).

The ExampleServer populates the endpoint list like this:
```java
        List<String> endpointAddresses = newArrayList();
        endpointAddresses.add(HostnameUtil.getHostname());
        endpointAddresses.addAll(HostnameUtil.getHostnames("0.0.0.0"));
```

but you're free to configure it however you want.

ctron, 2018-05-03:
Well I was more talking of the client side. In case the server does report the wrong information, or maybe it is not Milo at all :wink: … then you would need to have client side support for this.

kevinherron, 2018-05-03:
Just to give you an idea of how your application might handle this configuration, what ours does is: Endpoint Addresses is String setting, expecting a comma-separated list of hostnames to use, where a hostname surrounded by `<>` (so... `<myhost>`) is passed to `HostnameUtil.getHostnames()`, the `<hostname>` is a special case for `HostnameUtil.getHostname(), and anything else is used as it's received.

Given that `config.endpointAddressList` is a `List<String>`, then:
```kotlin
val endpointAddresses = config.endpointAddressList.flatMap {
    if ("<.+>".toRegex().matches(it)) {
        val hostname = it.removeSurrounding("<", ">")

        if (hostname == "hostname") {
            setOf(HostnameUtil.getHostname())
        } else {
            HostnameUtil.getHostnames(hostname)
        }
    } else {
        setOf(it)
    }
}
```

kevinherron, 2018-05-03:
Ah, on the client side... yeah, I guess it's not as easy as it could be. But I just point people here: https://stackoverflow.com/a/40555243/85927

kevinherron, 2018-05-03:
Sure. I think some kind of "builder" for EndpointDescription that lets you pass the original to set values and then optional set more on the builder would be okay. See how copying one of the config objects such as `OpcUaServerConfig` works.

Also, a helper for doing the endpoint URL is needed. It's not enough to just let you set a new endpoint URL, otherwise there's no need for this extra utility because you can just construct a new `EndpointDescription` the old fashioned way. Using URI has the drawbacks mentioned in the Stack Overflow post. Maybe a regex could be used in its place.

ctron, 2018-05-03:
well I think adding some kind of builder pattern would ensure that you won't run in too much issues when the constructor is changed: 

Something like:

~~~
EndpointDescription ed = …;
EndpointDescription ed2 = ed.newBuilder().hostname().build();
~~~

kevinherron, 2018-05-03:
On further thought, there's no reason for a builder because there's no valid reason to replace any of the other `EndpointDescription` fields.

How about just this:
```java
endpoint = EndpointUtil.replaceUrlHostname(endpoint, "new_host_name");
```

closed 2018-05-12 by kevinherron