eclipse-milo/milo #261: Speed

manijndl, 2018-05-05:
i am connecting to a OPC DA Schneider Server  with UA wrapper over it . i am using eclipse milo to connect but the response time to get tag values by function client.read() is very slow . please suggest something

kevinherron, 2018-05-05:
Sorry, you’ll have to look closer at either the wrapper or the server. The client is not responsible for slow response times from the server.

manijndl, 2018-05-05:
thanks kevin can i use use eclipse milo for OPC DA?

kevinherron, 2018-05-05:
No, Milo is only OPC UA.  I think there is a project called Utgard that has OPC classic support in Java but I’m not involved with it in any way.

manijndl, 2018-05-07:
one more question how can i get the changing tag values should i have to poll again and again to read the values or is there any better method to do that?

ctron, 2018-05-07:
@kevinherron is correct, there is a project called Utgard which can do OPC DA and OPC XML DA directly from pure Java. I am the original author of this library:

* https://search.maven.org/#artifactdetails%7Corg.openscada.utgard%7Corg.openscada.opc.dcom%7C1.5.0%7Cjar
* https://search.maven.org/#artifactdetails%7Corg.openscada.utgard%7Corg.openscada.opc.xmlda%7C1.5.0%7Cjar

manijndl, 2018-05-07:
@ctron i have trying from 2 days with Utgard but i am unable to connect with the OPC DA Server.

manijndl, 2018-05-07:
@ctron raised please check

closed 2018-05-07 by kevinherron