eclipse-milo/milo #263: can't find class in org.eclipse.milo.opcua.binaryschema.GenericBsdParser.java

MeowStone, 2018-05-07:
```
import org.eclipse.milo.opcua.binaryschema.parser.BsdParser;
import org.eclipse.milo.opcua.stack.core.serialization.codecs.OpcUaBinaryDataTypeCodec;
import org.opcfoundation.opcua.binaryschema.EnumeratedType;
import org.opcfoundation.opcua.binaryschema.StructuredType;

public class GenericBsdParser extends BsdParser {
 somecode.....
}
```
lack of class EnumeratedType and StructuredType

kevinherron, 2018-05-07:
Have you run `mvn install` from the base directory using JDK 8? These classes need to be generated.

kevinherron, 2018-05-07:
Alternatively, you can convince whatever IDE you use to run the `jaxb2:xjc` Maven plugin/goal.

kevinherron, 2018-05-08:
How does it fail? What's the error?

kevinherron, 2018-05-11:
Closing this. If you need further help on this topic use the mailing list.

closed 2018-05-11 by kevinherron