eclipse-milo/milo #266: no x509 certificate token policy found

point85, 2018-05-10:
Using the Milo 0.2.1 release, I am seeing a "no x509 certificate token policy found" issue when connecting to a Prosys OPC UA simulation server using Basic256Sha256/Sign&Encrpt with a self-signed X509 certificate from a Windows 10 machine with java 1.8.0_152 64 bits and an Eclipse development environment.  Note that an anonymous connection works fine.

Here are the steps that I have taken.

1.  Execute the openssl and keystore commands as documented in the milo examples:
openssl req -x509 -sha256 -newkey rsa:2048 -keyout out/privateKey.key -out out/certificate.crt -extensions v3_self_signed -config openssl.cnf
(openssl is version 1.1.0h 27 Mar 2018.  Note that the -config arg is slightly different than in the Milo example.  I also changed the URI in openssl.cnf for my project.  All other settings are the same.)

set RANDFILE=C:\tools\openssl\.rnd
openssl pkcs12 -export -name opcua -in out/certificate.crt -inkey out/privateKey.key > out/point85.p12

keytool -importkeystore -srckeystore out/point85.p12 -destkeystore out/point85.keystore -srcstoretype pkcs12 -alias opcua

2.  Add this certificate to the trusted ones on the Prosys server.

3. Read the keystore file with code taken from the KeyStoreLoader class in the standalone example and set the OpcUaClientConfigBuilder properties the same way:
...
			identityProvider = new X509IdentityProvider(loader.getClientCertificate(),
					loader.getClientKeyPair().getPrivate());
...
			OpcUaClientConfigBuilder configBuilder = new OpcUaClientConfigBuilder();

	configBuilder.setApplicationName(LocalizedText.english(APP_NAME)).setApplicationUri(APP_URI)
					.setEndpoint(endpointDescription).setIdentityProvider(identityProvider)
					.setRequestTimeout(uint(REQUEST_TIMEOUT));
...
			configBuilder.setCertificate(loader.getClientCertificate()).setKeyPair(loader.getClientKeyPair());
	
4.  On the first attempt, I got an "Illegal key size" exception:
UaException: status=Bad_SecurityChecksFailed, message=java.security.InvalidKeyException: Illegal key size
	at org.eclipse.milo.opcua.stack.core.channel.ChunkEncoder$SymmetricEncoder.getAndInitializeCipher(ChunkEncoder.java:409)
	at org.eclipse.milo.opcua.stack.core.channel.ChunkEncoder$AbstractEncoder.encode(ChunkEncoder.java:215)
	at org.eclipse.milo.opcua.stack.core.channel.ChunkEncoder.encode(ChunkEncoder.java:87)
	at org.eclipse.milo.opcua.stack.core.channel.ChunkEncoder.encodeSymmetric(ChunkEncoder.java:73)
	at org.eclipse.milo.opcua.stack.client.handlers.UaTcpClientMessageHandler.lambda$encode$5(UaTcpClientMessageHandler.java:344)
	at org.eclipse.milo.opcua.stack.core.channel.SerializationQueue.lambda$encode$0(SerializationQueue.java:55)
	at org.eclipse.milo.opcua.stack.core.util.ExecutionQueue$PollAndExecute.run(ExecutionQueue.java:107)
	at java.util.concurrent.Executors$RunnableAdapter.call(Executors.java:511)
	at java.util.concurrent.FutureTask.run(FutureTask.java:266)
	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1149)
	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:624)
	at java.lang.Thread.run(Thread.java:748)
Caused by: java.security.InvalidKeyException: Illegal key size
	at javax.crypto.Cipher.checkCryptoPerm(Cipher.java:1039)
	at javax.crypto.Cipher.implInit(Cipher.java:805)
	at javax.crypto.Cipher.chooseProvider(Cipher.java:864)
	at javax.crypto.Cipher.init(Cipher.java:1396)
	at javax.crypto.Cipher.init(Cipher.java:1327)
	at org.eclipse.milo.opcua.stack.core.channel.ChunkEncoder$SymmetricEncoder.getAndInitializeCipher(ChunkEncoder.java:403)
	... 11 more

5.  I then changed the jre/lib/security/java.security crypto.policy to unlimited:
# Please see the JCA documentation for additional information on these
# files and formats.
crypto.policy=unlimited

6.  Upon re-connecting, I get the "no x509 certificate token policy found" exception:
java.util.concurrent.ExecutionException: java.lang.Exception: no x509 certificate token policy found
2018-05-10 11:55:24.466 DEBUG SessionFsm - S(Activating) x E(ActivateSessionFailureEvent) = S'(Inactive)
	at java.util.concurrent.CompletableFuture.reportGet(CompletableFuture.java:357)
	at java.util.concurrent.CompletableFuture.get(CompletableFuture.java:1915)
	at org.point85.domain.opc.ua.UaOpcClient.connect(UaOpcClient.java:214)
	at org.point85.app.opc.ua.OpcUaController.connectToDataSource(OpcUaController.java:38)
	at org.point85.app.designer.DataSourceConnectionController$ConnectionService$1.call(DataSourceConnectionController.java:147)
	at org.point85.app.designer.DataSourceConnectionController$ConnectionService$1.call(DataSourceConnectionController.java:1)
	at javafx.concurrent.Task$TaskCallable.call(Task.java:1423)
	at java.util.concurrent.FutureTask.run(FutureTask.java:266)
	at javafx.concurrent.Service.lambda$null$493(Service.java:725)
	at java.security.AccessController.doPrivileged(Native Method)
	at javafx.concurrent.Service.lambda$executeTask$494(Service.java:724)
	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1149)
	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:624)
	at java.lang.Thread.run(Thread.java:748)
Caused by: java.lang.Exception: no x509 certificate token policy found
	at org.eclipse.milo.opcua.sdk.client.api.identity.X509IdentityProvider.lambda$getIdentityToken$1(X509IdentityProvider.java:56)
	at java.util.Optional.orElseThrow(Optional.java:290)
	at org.eclipse.milo.opcua.sdk.client.api.identity.X509IdentityProvider.getIdentityToken(X509IdentityProvider.java:56)
	at org.eclipse.milo.opcua.sdk.client.session.states.AbstractSessionState.lambda$activateSession$7(AbstractSessionState.java:231)
	at java.util.concurrent.CompletableFuture.uniComposeStage(CompletableFuture.java:981)
	at java.util.concurrent.CompletableFuture.thenCompose(CompletableFuture.java:2124)
	at org.eclipse.milo.opcua.sdk.client.session.states.AbstractSessionState.activateSession(AbstractSessionState.java:220)
	at org.eclipse.milo.opcua.sdk.client.session.states.AbstractSessionState.lambda$activateSessionAsync$5(AbstractSessionState.java:207)

7.  If I try a connection to the Unified Automation Ansi C demo server with Basic256Sha256/Sign&Encrypt and the same certificate and key pair, I get a different exception:

java.util.concurrent.ExecutionException: UaServiceFaultException: status=Bad_IdentityTokenRejected, message=The user identity token is valid but the server has rejected it.
2018-05-10 13:07:40.978 DEBUG SessionFsm - S(Activating) x E(ActivateSessionFailureEvent) = S'(Inactive)
	at java.util.concurrent.CompletableFuture.reportGet(CompletableFuture.java:357)
	at java.util.concurrent.CompletableFuture.get(CompletableFuture.java:1915)
	at org.point85.domain.opc.ua.UaOpcClient.connect(UaOpcClient.java:214)
	at org.point85.app.opc.ua.OpcUaController.connectToDataSource(OpcUaController.java:38)
	at org.point85.app.designer.DataSourceConnectionController$ConnectionService$1.call(DataSourceConnectionController.java:147)
	at org.point85.app.designer.DataSourceConnectionController$ConnectionService$1.call(DataSourceConnectionController.java:1)
	at javafx.concurrent.Task$TaskCallable.call(Task.java:1423)
	at java.util.concurrent.FutureTask.run(FutureTask.java:266)
	at javafx.concurrent.Service.lambda$null$493(Service.java:725)
	at java.security.AccessController.doPrivileged(Native Method)
	at javafx.concurrent.Service.lambda$executeTask$494(Service.java:724)
	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1149)
	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:624)
	at java.lang.Thread.run(Thread.java:748)
Caused by: UaServiceFaultException: status=Bad_IdentityTokenRejected, message=The user identity token is valid but the server has rejected it.
	at org.eclipse.milo.opcua.stack.client.UaTcpStackClient.lambda$receiveResponse$14(UaTcpStackClient.java:358)
	at org.eclipse.milo.opcua.stack.core.util.ExecutionQueue$PollAndExecute.run(ExecutionQueue.java:107)
	at java.util.concurrent.Executors$RunnableAdapter.call(Executors.java:511)
	at java.util.concurrent.FutureTask.run(FutureTask.java:266)
	... 3 more

It would appear that the X509 certificate is not correct, but I do not know what is wrong with it.

Thanks for your help!

kevinherron, 2018-05-10:
The error connecting to Prosys would suggest you're not selecting an endpoint that user a `UserTokenPolicy` with `UserTokenType.Certificate` on it. Would have to see your code or some logging about what endpoint you choose to guess further.

The error connecting to the Ansi C demo server... have you told it (somehow, I don't know how) to actually trust your user certificate? This might be your identity being rejected because it's not trusted.

point85, 2018-05-10:
My code searches for an exact match to get an endpoint:
// get the server's endpoints
EndpointDescription[] endpointDescriptions = UaTcpStackClient.getEndpoints(
endpointUrl).get();

// security settings
String policyUri = source.getSecurityPolicy().getSecurityPolicyUri();
MessageSecurityMode messageSecurityMode = source.getMessageSecurityMode();

logger.info("Configured policy: " + policyUri + ", mode: " +
messageSecurityMode);

EndpointDescription endpointDescription = null;

for (EndpointDescription description : endpointDescriptions) {
String uri = description.getSecurityPolicyUri();
MessageSecurityMode mode = description.getSecurityMode();

logger.info("Checking URL: " + description.getEndpointUrl() + ",  Policy: "
+ uri + ", Mode: " + mode);

if (uri.equals(policyUri) && mode.equals(messageSecurityMode)) {
endpointDescription = description;
break;
}
}

if (endpointDescription == null) {
String msg = "Unable to find a matching endpoint for security policy " +
policyUri + " and mode "
+ messageSecurityMode;
logger.error(msg);
throw new Exception(msg);
}

logger.info("Using endpoint: {} [{}, {}]", endpointDescription.
getEndpointUrl(),
endpointDescription.getSecurityPolicyUri(), endpointDescription.
getSecurityMode());

I have also tried the "chooseEndpoint()" method in the standalone client
example, and it results in the same exception as my code.

Here's a log with Milo set to "trace".  "UaOpcClient" is my class that
calls the OpcUaClient's connect() method

2018-05-10 16:12:15.808 INFO  UaOpcClient - Connecting to:
opc.tcp://Kent-acer:53530/OPCUA/SimulationServer
2018-05-10 16:12:20.639 INFO  X509KeyStoreLoader - Reading keystore
point85.keystore
2018-05-10 16:12:20.641 INFO  X509KeyStoreLoader - Loading KeyStore at
C:\dev\OEE-Domain\target\classes\org\point85\domain\opc\ua\point85.keystore
2018-05-10 16:12:21.021 INFO  X509KeyStoreLoader - Loaded KeyStore at
C:\dev\OEE-Domain\target\classes\org\point85\domain\opc\ua\point85.keystore
2018-05-10 16:12:21.023 INFO  UaOpcClient - Identity provider:
X509IdentityProvider
2018-05-10 16:12:22.697 DEBUG ClientChannelManager - connect(),
currentState=NotConnected
2018-05-10 16:12:22.699 DEBUG ClientChannelManager - connect() while
NotConnected
java.lang.Exception
at org.eclipse.milo.opcua.stack.client.ClientChannelManager.
connect(ClientChannelManager.java:67)
at org.eclipse.milo.opcua.stack.client.UaTcpStackClient.
connect(UaTcpStackClient.java:127)
at org.eclipse.milo.opcua.stack.client.UaTcpStackClient.
getEndpoints(UaTcpStackClient.java:568)
at org.point85.domain.opc.ua.UaOpcClient.connect(UaOpcClient.java:165)
at org.point85.app.opc.ua.OpcUaController.connectToDataSource(
OpcUaController.java:38)
[... 173 lines cut ...]
ThreadPoolExecutor.java:624)
at java.lang.Thread.run(Thread.java:748)
Caused by: java.lang.Exception: no x509 certificate token policy found
at org.eclipse.milo.opcua.sdk.client.api.identity.
X509IdentityProvider.lambda$getIdentityToken$1(X509IdentityProvider.java:56)
at java.util.Optional.orElseThrow(Optional.java:290)
at org.eclipse.milo.opcua.sdk.client.api.identity.X509IdentityProvider.
getIdentityToken(X509IdentityProvider.java:56)
at org.eclipse.milo.opcua.sdk.client.session.states.
AbstractSessionState.lambda$activateSession$7(AbstractSessionState.java:231)
at java.util.concurrent.CompletableFuture.uniComposeStage(
CompletableFuture.java:981)
at java.util.concurrent.CompletableFuture.thenCompose(
CompletableFuture.java:2124)
at org.eclipse.milo.opcua.sdk.client.session.states.AbstractSessionState.
activateSession(AbstractSessionState.java:220)
at org.eclipse.milo.opcua.sdk.client.session.states.
AbstractSessionState.lambda$activateSessionAsync$5(
AbstractSessionState.java:207)
... 3 more

Regarding the Unified Automation C (and C++) demo servers, after the
certificate is rejected I move it to the trusted folder and restart the
server.  By the way, there are no messages on any of the server consoles
when the exceptions happen.

kevinherron, 2018-05-11:
Your code only looks at security policy and message security mode, it doesn't make sure that the endpoint actually has a `UserTokenPolicy` of type `UserTokenType.Certificate` in the `EndpointDescription`s `userIdentityTokens` field.

I found that the Prosys server only starts returning endpoints with these policies once you've gone to the Users tab, enabled Certificate type, then restarted it.

As for the Unified Automation server - are you sure you're not talking about moving the application instance certificate from rejected to trusted? The certificate you're trying to use as an identity is not the same thing as the application instance certificate. Did you open the certificate with your OS's certificate viewer and make sure it's the new identity cert you created?

point85, 2018-05-11:
I wrote the code to check the UserTokenType for my chosen endpoint, and the
Certificate type is there.

Enabling the certificate types in the Prosys server User tab and restarting
it did make a difference.  Now it behaves the same way as the Unified
Automation C and C++ demo servers, namely it rejects a valid user identity
token:
2018-05-10 17:41:50.400 DEBUG SessionFsm - S(Creating) x
E(CreateSessionSuccessEvent) = S'(Activating)
2018-05-10 17:41:50.429 DEBUG SessionFsm - Sending ActivateSessionRequest,
secureChannelId=5, channel=[id: 0xccb84ffc, L:/192.168.0.8:8775 -
R:LenovoE555/192.168.0.8:53530]...
2018-05-10 17:41:50.429 TRACE ClientChannelManager - getChannel(),
currentState=Connected
2018-05-10 17:41:50.429 TRACE UaTcpStackClient - writeAndFlush succeeded
for request=ActivateSessionRequest, requestHandle=1
2018-05-10 17:41:50.462 DEBUG UaTcpStackClient - Received ServiceFault
requestHandle=1, result=StatusCode{name=Bad_IdentityTokenRejected,
value=0x80210000, quality=bad}
2018-05-10 17:41:50.464 DEBUG SessionFsm - S(Activating) x
E(ActivateSessionFailureEvent) = S'(Inactive)
java.util.concurrent.ExecutionException: UaServiceFaultException:
status=Bad_IdentityTokenRejected, message=The user identity token is valid
but the server has rejected it.
at
java.util.concurrent.CompletableFuture.reportGet(CompletableFuture.java:357)
at java.util.concurrent.CompletableFuture.get(CompletableFuture.java:1915)
at org.point85.domain.opc.ua.UaOpcClient.connect(UaOpcClient.java:231)
at
org.point85.app.opc.ua.OpcUaController.connectToDataSource(OpcUaController.java:38)
at
org.point85.app.designer.DataSourceConnectionController$ConnectionService$1.call(DataSourceConnectionController.java:147)
at
org.point85.app.designer.DataSourceConnectionController$ConnectionService$1.call(DataSourceConnectionController.java:1)
at javafx.concurrent.Task$TaskCallable.call(Task.java:1423)
at java.util.concurrent.FutureTask.run(FutureTask.java:266)
at javafx.concurrent.Service.lambda$null$493(Service.java:725)
at java.security.AccessController.doPrivileged(Native Method)
at javafx.concurrent.Service.lambda$executeTask$494(Service.java:724)
at
java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1149)
at
java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:624)
at java.lang.Thread.run(Thread.java:748)
Caused by: UaServiceFaultException: status=Bad_IdentityTokenRejected,
message=The user identity token is valid but the server has rejected it.
at
org.eclipse.milo.opcua.stack.client.UaTcpStackClient.lambda$receiveResponse$14(UaTcpStackClient.java:358)
at
org.eclipse.milo.opcua.stack.core.util.ExecutionQueue$PollAndExecute.run(ExecutionQueue.java:107)
at java.util.concurrent.Executors$RunnableAdapter.call(Executors.java:511)
at java.util.concurrent.FutureTask.run(FutureTask.java:266)
... 3 more

I am attaching a screen capture of the Prosys dialog and the Windows
certificate viewer.

kevinherron, 2018-05-11:
The Prosys example works for me once I move the identity certificate from `~/.prosys/SimulationServer/USERS_PKI/CA/rejected` to `~/.prosys/SimulationServer/USERS_PKI/CA/trusted`

I think the issue is likely the same for your Unified Automation demo server - the identity certificate is not trusted.

point85, 2018-05-11:
It works!  I spent two days on trying to get this to work.  Thanks a bunch
Kevin.

kevinherron, 2018-05-11:
No problem. 

If it wasn't a clear takeaway, when selecting endpoints, you need to treat the presence of your desired `UserTokenPolicy` just as importantly as the presence of the security profile and message security mode you're looking for. Not all endpoints will have all tokens, and if you blindly choose the first one that meets the other criteria you might not be able to connect with the authentication mechanism you want.

closed 2018-05-11 by kevinherron