eclipse-milo/milo #268: Difference

manijndl, 2018-05-14:
Method 1:- VariableNodevarnode=client.getAddressSpace().createVariableNode(node)
DataValue value = varnode.readValue().get();
System.out.println("" + value.getValue().getValue());

Method2 :-
ReadValueId readValueId = new ReadValueId(NodeId.parse(browseId), AttributeId.DataType.uid(), null,null);
ReadResponse response=client.read(2147483647,TimestampsToReturn.Both,Arrays.asList(readValueId)).get();

what is the difference between these two methods both are reading value?

kevinherron, 2018-05-14:
Both read the value. They're just different abstractions.

One leaves you with a `VariableNode` that represents the node and you can use to read/write other attributes.

The other is the lower-level call using the `OpcUaClient` object directly. This gives you more control, and would allow you to e.g. read attributes in bulk.

closed 2018-05-14 by kevinherron

manijndl, 2018-05-15:
which method is speed efficient if only read operation is required ?

and is possible to get only value returned by method 1? Example only 12 
instead of this {Variant{value=12}  ?

kevinherron, 2018-05-15:
Calling `client.read()` directly is more efficient, but you're always going to have to "unwrap" the value to get at it.