eclipse-milo/milo #270: Allow connection exceptions/errors to be notified outside of inner client

simone201, 2018-05-16:
## Versions
Eclipse Milo SDK Client => 0.2.1

## Issue
The client is missing a way to handle or get notified of any connection issue such as lost connection, closed or any error that may require a restart of the whole client. As now exceptions are handled only internally via the "exceptionCaught" method which just closes the connection and context leaving the developer without any clue of the error apart from a stacktrace.

There's also an included API in OpcUaClient which is "addFaultListener" that as now does pretty much nothing, but may be used also to handle these connection errors.

This is a really huge lack to consider this library at least "usable".

kevinherron, 2018-05-16:
Try `OpcUaClient#addSessionActivityListener()`.

I'm not sure what other internal exceptions you're referring to, especially ones that may require a restart of the client.

kevinherron, 2018-05-18:
@simone201 can you provide more details?

kevinherron, 2018-05-23:
Feel free to re-open this or use the mailing list of you'd like to elaborate.

closed 2018-05-23 by kevinherron