eclipse-milo/milo #275: Run the example

gkrananth, 2018-05-19:
I successfully compiled through eclipse maven project.  How do compile the example ReadExample.java from command prompt.

kevinherron, 2018-05-21:
You can either run the straight out of your IDE or from a terminal with something like: `mvn exec:java -D"exec.mainClass"="org.eclipse.milo.examples.client.ReadExample"` from the `client-examples` directory.

closed 2018-05-21 by kevinherron

kevinherron, 2018-05-23:
`-DskipTests`

This is not Milo-specific, just basic Maven stuff.

kevinherron, 2018-05-23:
You should always start by running `mvn clean install` from the project root. Skip tests if you need to, but it shouldn't be necessary.

gkrananth, 2018-05-23:
mvn clean install got success. Am getting error on exec command

mvn exec:java
-D"exec.mainClass"="org.eclipse.milo.examples.client.ReadNodeExample"

Best Regards

G Anantharaj

kevinherron, 2018-05-23:
In a fresh Ubuntu VM I ran `mvn clean install -DskipTests` at the project root.

Once this finished, I changed directory to `./milo-examples/client-examples` and ran `mvn exec:java -D"exec.mainClass"="org.eclipse.milo.examples.client.ReadNodeExample"`. 

The example ran without error.

Make sure you're using JDK8 and maybe try running the examples out of an IDE instead if you can't figure out what's wrong. I'm not sure how else to help you.

gkrananth, 2018-05-24:
thank you, it is working.

gkrananth, 2018-06-15:
Hi

I am facing some error after adding sqllite code

[INFO] Starting audit...
[ERROR]
/home/ipc-03/Desktop/milo-master/milo-examples/client-examples/src/main/java/org/eclipse/milo/examples/client/ReadCustomDataTypeNodeExample.java:30:
Import statement for 'java.sql.Connection' is in the wrong order. Should be
in the 'STANDARD_JAVA_PACKAGE' group, expecting not assigned imports on
this line. [CustomImportOrder]

How do i remove checkstyle checking?
I am only changing one file in client examples source how do i compile only
that changes.

Please let me know

Thanks

G Anantharaj

kevinherron, 2018-06-15:
`-Dcheckstyle.skip=true`

gkrananth, 2018-06-15:
thank you very much.