eclipse-milo/milo #276: Reading history OPC_UA

manijndl, 2018-05-23:
hi can you please help me in reading history data from OPC_UA server using eclipse milo?

client.historyRead(historyReadDetails, timestampsToReturn, releaseContinuationPoints, nodesToRead)

wat to add in historyReadDetails and releaseContinuationPoints?
how much history data is available in OPC_Ua server ? is it going to read from the beginning of time?

manijndl, 2018-05-23:
[Code.txt](https://github.com/eclipse/milo/files/2029777/Code.txt)
[response.txt](https://github.com/eclipse/milo/files/2029779/response.txt)

Unable to Read the history Data i have attached the codefile and reponse i got

kevinherron, 2018-05-23:
`Bad_NodeIdUnknown` comes from the server, not sure there's anything else I can suggest other than making sure you absolutely have the correct NodeId.

Reading history is going to require that you look at Part 4 and probably Part 11 of the OPC UA specifications to get a better idea how the history services work. 

Milo server does not implement any history so you'll also need another 3rd party server that *does* support it (looks like you might already have this).

manijndl, 2018-05-24:
i am not using any third party library and node Id is correct

milo has HistoryRead method will it going to work for history read?

kevinherron, 2018-05-24:
Yes, the client supports reading history from servers that support it as well.

kevinherron, 2018-05-24:
I just added an example of HistoryRead that runs against Prosys example server: https://github.com/eclipse/milo/commit/c994e584bb7bec30e022de661e4e72329b400c23#diff-3513fee29042bdae2a75c26f5fb45988

closed 2018-05-24 by kevinherron

manijndl, 2018-05-25:
is history code varies from server to server? suppose i have written a code for prosys then i have to my code for matricon or schnieder?

manijndl, 2018-05-25:
i have tried this example code 

getting response

HistoryReadResult{StatusCode=StatusCode{name=Bad_NodeIdUnknown, value=0x80340000, quality=bad}, ContinuationPoint=ByteString{bytes=null}, HistoryData=ExtensionObject{encoded=null, encodingTypeId=NodeId{ns=0, id=0}}}

I am very sure NodeId is correct because i am reading live data of this nodeId using eclipse Milo please help in this

kevinherron, 2018-05-25:
The code will vary in what NodeId and details you request but not much else.

You might have to contact your server vendor for help. The StatusCode comes from the server.

manijndl, 2018-05-28:
how do i know the opc server supports history read?

manijndl, 2018-06-11:
Error running client example: UaServiceFaultException: status=Bad_InvalidTimestampArgument, message=The defined timestamp to return was invalid.
java.util.concurrent.ExecutionException: UaServiceFaultException: status=Bad_InvalidTimestampArgument, message=The defined timestamp to return was invalid.

getting this exception while running history example

kevinherron, 2018-06-11:
Again, the StatusCode comes from the server. I'm not sure what it means in this case. Try changing the dates in the requested range or the `TimestampsToReturn` requested.