eclipse-milo/milo #285: Browse to a node

hbbend, 2018-06-05:
Hi,
is there an instruction to browse to a folder in servers address space then select this one, and browse it.
using MILO OPC UA Client
thank you

kevinherron, 2018-06-05:
Have you looked at the [Browse Example](https://github.com/eclipse/milo/blob/23aa45711afec83d1774c21f2bcd5b4606bebfc3/milo-examples/client-examples/src/main/java/org/eclipse/milo/examples/client/BrowseExample.java) yet?

hbbend, 2018-06-06:
I did that, it's successfull with prosys simulation opc server, I browse, read, write and subscribe, without any problem. However, for PLC opc server, the variables are stored in : object/PLC_1/Memory/VariableNode, and they appear when i browse the Rootfolder, but when I read, i tell me that the node does not exist.
i did read the variable node using another client, by browsing to PLC_1 then selecting PLC_1 and browsing content, then selecting Memory node, then browsing content and reading Variablenode successfuly.
But I have to do it with Milo client?
thank you
Best Regards

kevinherron, 2018-06-06:
The read will work as long as you specify the correct NodeId. You would have to discover that NodeId during your browse process if you don’t know it ahead of time. Maybe you can post some of your code you’re trying to get working.

hbbend, 2018-06-06:
Now I could browse into the folder Memory containing the Variable node, however I have to read it's value, for that I tryed this one: (but never find the node to read the value, even if the string matches)
...
      		MemeNodeId = TestNodeId2;//nodePLC.getNodeId().get();
      		MemNodes = client.getAddressSpace().browse(MemeNodeId).get();
      	//	nodeVarLue =	client.getAddressSpace().getVariableNode(NodeIdVariable).get();
      	  for (Node nodeMem : MemNodes) {
      		TestNodeId3 = nodeMem.getNodeId().get();
      	//	verif = nodeMem.readNodeId().get();
          	logger.info("{} NodeMem={}", TestNodeId3);

      	String st1 = TestNodeId3.getIdentifier().toString();
      	String st2 = "\""+NodeIdVariable.getIdentifier().toString()+"\"";
      	
      //	st1 =trimDoubleQuotes(st1);
      //	String st2 = NodeIdVariable.getIdentifier().toString();
      	System.out.println("*********** ST1 : "+st1);
      	System.out.println("*********** ST2 : "+st2);

        	if (st1 == st2) {
        		
               	//****************************************************       		           
        		
          	      nodeVarLue = client.getAddressSpace().createVariableNode(TestNodeId3);
          	       logger.info("DataType={}", nodeVarLue.getDataType().get());
          	      logger.info("DataType={}", nodeVarLue.getBrowseName().get());
          	      
          	       DataValue value1 = nodeVarLue.readValue().get();	
          	       Integer ValeurLue = (Integer) value1.getValue().getValue();
          	       logger.info("ValeurLue hiya {}", ValeurLue);
              break;  	//**************************************************          	       
        	}
        	}      	    	        	       
      	 }	  
  }

kevinherron, 2018-06-08:
One immediate problem that stands out is that you’re doing the string equality wrong. You have to use the .equals() method in Java.

closed 2018-06-08 by kevinherron

hbbend, 2018-06-12:
Hi Kevin,
thank you, it worked !
I can now read the node by browsing into it, then read. 
However, when i substitute the reading by a write, the bad news is that i can not write to the node,
please find below the code:
    	       int v = 22;
    	            Variant vv = new Variant(v);
    	       
    	       DataValue VV = new DataValue(vv, null, null);
    	       NodeId nodeId1 = new NodeId(3, "Consigne_MOT_Vitesse _1");     
    	      client.writeValue(NodeId1, VV);
   	      nodeVarLue = client.getAddressSpace().createVariableNode(TestNodeId);   
   	      value1 = nodeVarLue.readValue().get();	
    	      ValeurLue = value1.getValue().getValue();            	    	            	               	    	            	       
    	      System.out.println("The new value of the variable : "+ NodeIdVariable.getIdentifier() + " is : "+ ValeurLue);           	    	            	                   	    	            	     

It executes the code then returns the old value of the variable without changing it!!?

Kbhargav04, 2019-08-19:
Hi Kevin,

Im trying to read from GE Cimplicity OPC UA.
i am able to read some Tags. But some of the tags return me bad nodeId unknown.
i am using node Id as 3 for reading the tags in the project.