eclipse-milo/milo #286: can not write to a node

hbbend, 2018-06-12:
Hi Kevin,
My problem is, when i connect to a PLC OPC UA server I can browse, but i can not read nor write. (bad nod Id Unknown, this node does not exist in the servers address space).
""...it works successfully with other servers..."".

I solved the problem of reading by browsing to the node then read it. (see #285 )
however when i try the same thing with writing, I do not get any exception, but it the value attribute does not change, it simply don't write.
would you help me with this, thank you
B/R

kevinherron, 2018-06-12:
You're not waiting for the write to finish in your example. All calls are async.

Try this instead: `client.writeValue(NodeId1, VV).get();`

hbbend, 2018-06-12:
same result, node value still the same.

please find below the code 

	private Object HbrowseWrite(OpcUaClient client, NodeId browseRoot, NodeId NodeIdVariable) {
		
		Object ValeurLue = null;
		Object resultat = null;
	    int v = 22;
        Variant vv = new Variant(v);  
        DataValue VV = new DataValue(vv, null, null);
		
		try {
        	
            NodeId TestNodeId = null;
             
            VariableNode nodeVarLue = null;
            DataValue value1;
            
            List<Node> nodes = client.getAddressSpace().browse(browseRoot).get();

            for (Node node : nodes) {
            	TestNodeId = node.getNodeId().get();
            
            	 if (CompareNodeId(TestNodeId, NodeIdVariable))
            	 {
          		            	    
            		// new WriteValue(TestNodeId, uint(13), null, VV);   
    	     // client.writeValue(TestNodeId, VV);
    	      client.writeValue(TestNodeId, VV).get();
    	      nodeVarLue = client.getAddressSpace().createVariableNode(TestNodeId);   
   	      value1 = nodeVarLue.readValue().get();	
    	      ValeurLue = value1.getValue().getValue();            	    	            	               	    	            	       
    	      System.out.println("La nouvelle valeur de la variable : "+ NodeIdVariable.getIdentifier() + " est : "+ ValeurLue);           	    	            	                   	    	            	     
            	    
            	    
            	    break; 
            	 } 
                // recursively browse to children
                resultat = HbrowseWrite(client, node.getNodeId().get(), NodeIdVariable);
                if (resultat != null) {
                	ValeurLue = resultat;
                	break;
                }
                
            }
        } catch (InterruptedException | ExecutionException e) {
            logger.error("Browsing nodeId={} failed: {}", browseRoot, e.getMessage(), e);
        }
		
		return ValeurLue;
    }

I am using the browse with this server to read and write, because it does not find the node when simply read and write, as described above.
Thanks

hbbend, 2018-06-12:
private boolean CompareNodeId(NodeId Node1, NodeId Node2){
		boolean result = false;
		if ((trimDoubleQuotes(Node1.getIdentifier().toString()).equals(Node2.getIdentifier().toString()) ) &
				(Node1.getNamespaceIndex().equals(Node1.getNamespaceIndex()))) result = true;
   	 {
   	 
   	 }
		
		return result;
		
	}
	
	public static String trimDoubleQuotes(String text) {
	    int textLength = text.length();

	    if (textLength >= 2 && text.charAt(0) == '"' && text.charAt(textLength - 1) == '"') {
	      return text.substring(1, textLength - 1);
	    }

	    return text;
	  }

kevinherron, 2018-06-12:
Check the StatusCode returned by the server when you write. If it's Good, but the value doesn't change, the issue is not on the client side and you'll have to troubleshoot the server somehow.

hbbend, 2018-06-12:
is there an instruction for that, or where to look?
thanks

kevinherron, 2018-06-12:
When you call `client.writeValue(TestNodeId, VV).get()` the return value is a `StatusCode`...

hbbend, 2018-06-12:
ah ok :))

hbbend, 2018-06-12:
StatusCode{name=Bad_TypeMismatch, value=0x80740000, quality=bad}

hbbend, 2018-06-12:
I tried with UInteger v = uint(22); same result

hbbend, 2018-06-12:
[image]

kevinherron, 2018-06-12:
`Int16` is a Java `Short`, so make sure the value you're writing is cast to Short before you write it.

hbbend, 2018-06-12:
it seems so simple dut, it does not acespt cast to Ushort, and  i don(t find how to import to Milo maven project a library who does that

kevinherron, 2018-06-12:
I said `Short`, not `UShort`.

Something like: 
```java
Variant v = new Variant((short) 42);
DataValue dv = DataValue.valueOnly(v); 
```

hbbend, 2018-06-12:
ok, i did it, but i got an exception:

`15:41:06.397 [main] ERROR org.eclipse.milo.Client.ClientRunner - Error running example: Variant cannot contain Variant
java.lang.IllegalArgumentException: Variant cannot contain Variant`

hbbend, 2018-06-12:
sorry, one moment please

hbbend, 2018-06-12:
ouf, its working !

hbbend, 2018-06-12:
Thanks kevin, 
by the way, how can i run my project in another pc, because after i imported from existing maven projects, several projects are open in my workspace. And I want to pack them as before to use in another pc

closed 2018-06-12 by hbbend