eclipse-milo/milo #289: How to handle a "The session id is not valid." exception?

point85, 2018-06-13:
I am getting a "The session id is not valid." exception when reading the value of a node after the session has been unused for a few minutes.  The stack trace is below.  

What is the proper exception handling for this case?  I see that there is a OpcUaClientConfigBuilder.setSessionTimeout() method.  Should this be used?  If so, what are the units on the timeout value? Is it milli-seconds?  What is the default value?

Or, should a task periodically read the server's timestamp to keep the session alive?  If so, how often should this be done?

Thanks for the help!

2018-06-13 10:41:51.045 DEBUG UaTcpStackClient - Received ServiceFault requestHandle=26, result=StatusCode{name=Bad_SessionClosed, value=0x80260000, quality=bad}
2018-06-13 10:41:51.048 DEBUG OpcUaClient - Notifying 1 ServiceFaultListeners
2018-06-13 10:41:51.063 DEBUG SessionFsm$FaultListener - ServiceFault: StatusCode{name=Bad_SessionClosed, value=0x80260000, quality=bad}
2018-06-13 10:41:51.075 TRACE ClientChannelManager - getChannel(), currentState=Connected
2018-06-13 10:41:51.085 DEBUG ClientChannelManager - Scheduling reconnect for +0 seconds...
2018-06-13 10:41:51.099 DEBUG SessionFsm - S(Active) x E(ServiceFaultEvent) = S'(Reactivating)
2018-06-13 10:41:51.102 DEBUG SessionFsm - notifySessionInactive()
2018-06-13 10:41:51.108 TRACE ClientChannelManager - getChannel(), currentState=Reconnecting
2018-06-13 10:41:51.134 DEBUG SessionFsm - notifying 1 listeners...
2018-06-13 10:41:51.201 DEBUG ClientChannelManager - 0 seconds elapsed; reconnecting...
2018-06-13 10:41:51.232 DEBUG UaTcpClientAcknowledgeHandler - Sent Hello message on channel=[id: 0xbf0ac980, L:/192.168.0.8:3838 - R:/192.168.0.13:53530].
2018-06-13 10:41:51.235 DEBUG UaTcpClientAcknowledgeHandler - Received Acknowledge message on channel=[id: 0xbf0ac980, L:/192.168.0.8:3838 - R:/192.168.0.13:53530].
2018-06-13 10:41:51.236 DEBUG UaTcpClientMessageHandler - OpenSecureChannel timeout scheduled for +5s
2018-06-13 10:41:51.238 DEBUG UaTcpClientMessageHandler - Sent OpenSecureChannelRequest (Issue, id=0, currentToken=-1, previousToken=-1).
2018-06-13 10:41:51.241 DEBUG UaTcpClientMessageHandler - OpenSecureChannel timeout canceled
2018-06-13 10:41:51.242 DEBUG UaTcpClientMessageHandler - Received OpenSecureChannelResponse.
2018-06-13 10:41:51.243 DEBUG UaTcpClientMessageHandler - SecureChannel id=219, currentTokenId=1, previousTokenId=-1, lifetime=3600000ms, createdAt=DateTime{utcTime=131733853064250000, javaDate=Wed Jun 13 10:41:46 PDT 2018}
2018-06-13 10:41:51.243 DEBUG UaTcpClientMessageHandler - 0 message(s) queued before handshake completed; sending now.
2018-06-13 10:41:51.244 DEBUG ClientChannelManager - Channel bootstrap succeeded: localAddress=/192.168.0.8:3838, remoteAddress=/192.168.0.13:53530
2018-06-13 10:41:51.245 DEBUG SessionFsm - Sending (re)ActivateSessionRequest, secureChannelId=219, channel=[id: 0xbf0ac980, L:/192.168.0.8:3838 - R:/192.168.0.13:53530]...
2018-06-13 10:41:51.246 TRACE ClientChannelManager - getChannel(), currentState=Reconnecting
2018-06-13 10:41:51.252 TRACE UaTcpStackClient - writeAndFlush succeeded for request=ActivateSessionRequest, requestHandle=27
2018-06-13 10:41:51.252 DEBUG ClientChannelManager - Reconnect succeeded, channelId=219
2018-06-13 10:41:51.258 DEBUG UaTcpStackClient - Received ServiceFault requestHandle=27, result=StatusCode{name=Bad_SessionIdInvalid, value=0x80250000, quality=bad}
2018-06-13 10:41:51.259 DEBUG SessionFsm - (re)ActivateSession failed: OpcUaSession{sessionId=NodeId{ns=1, id=0d485404-757c-405e-8dad-f8dd7b72500e}, sessionName=UaSession:Point85 OEE OPC UA Client:1528911276936}
java.util.concurrent.CompletionException: UaServiceFaultException: status=Bad_SessionIdInvalid, message=The session id is not valid.
	at java.util.concurrent.CompletableFuture.encodeRelay(CompletableFuture.java:326)
	at java.util.concurrent.CompletableFuture.completeRelay(CompletableFuture.java:338)
	at java.util.concurrent.CompletableFuture.uniRelay(CompletableFuture.java:911)
	at java.util.concurrent.CompletableFuture$UniRelay.tryFire(CompletableFuture.java:899)
	at java.util.concurrent.CompletableFuture.postComplete(CompletableFuture.java:474)
	at java.util.concurrent.CompletableFuture.completeExceptionally(CompletableFuture.java:1977)
	at org.eclipse.milo.opcua.stack.client.UaTcpStackClient.lambda$receiveResponse$14(UaTcpStackClient.java:358)
	at org.eclipse.milo.opcua.stack.core.util.ExecutionQueue$PollAndExecute.run(ExecutionQueue.java:107)
	at java.util.concurrent.Executors$RunnableAdapter.call(Executors.java:511)
	at java.util.concurrent.FutureTask.run(FutureTask.java:266)
	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1149)
	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:624)
	at java.lang.Thread.run(Thread.java:748)
Caused by: UaServiceFaultException: status=Bad_SessionIdInvalid, message=The session id is not valid.
	... 7 more
2018-06-13 10:41:51.264 DEBUG SessionFsm - S(Reactivating) x E(ReactivateFailureEvent) = S'(Recreating)
2018-06-13 10:41:51.265 DEBUG SessionFsm - Sending CreateSessionRequest...
2018-06-13 10:41:51.265 TRACE ClientChannelManager - getChannel(), currentState=Connected
2018-06-13 10:41:51.266 TRACE UaTcpStackClient - writeAndFlush succeeded for request=CreateSessionRequest, requestHandle=28
2018-06-13 10:41:51.485 DEBUG SessionFsm - CreateSession succeeded: NodeId{ns=1, id=4211cacb-a466-4aae-93ff-38d0e3adf84d}

kevinherron, 2018-06-13:
> I see that there is a OpcUaClientConfigBuilder.setSessionTimeout()
method. Should this be used? If so, what are the units on the timeout
value? Is it milli-seconds? What is the default value?

You can set this to something greater, but that would just be an interim
fix. The default value is 120,000ms. I see the getters on OpcUaClientConfig
are all documented (including units) but the setters on the builders are
all lacking java docs right now... I'll have to fix that.

The real fix would be to have your client issue some kind of periodic
keep-alive to keep the session alive. Reading the CurrentTime and/or State
variables is suggested.

Coincidentally... this won't be necessary in 0.3. I've implemented a
keep-alive mechanism as part of the FSM that manages the session in
preparation for implementing the HTTP/HTTPS transports. These transports
are connection-less and can't rely on getting notified the socket
connection was broken (which doesn't work well anyway, mercy of the OS and
its TCP keep alive settings, etc...) Instead, a periodic read to the
ServerState variable is made and after some number of consecutive failures
the connection will be deemed broken/inactive.

point85, 2018-06-13:
Thanks Kevin.  I appreciate the quick response!

kevinherron, 2018-06-13:
👍

closed 2018-06-13 by kevinherron

shahalamansari, 2020-06-29:
@kevinherron  "The real fix would be to have your client issue some kind of periodic
keep-alive to keep the session alive. Reading the CurrentTime and/or State
variables is suggested." Can you please provide an example as to how to periodically schedule keep - alive  with SubscribeOPCUA nodes?

kevinherron, 2020-06-29:
@shahalamansari this hasn't been necessary for a long time. The client SDK automatically reads the server's State variable as a keep alive.

shahalamansari, 2020-06-29:
@kevinherron I am using eclipse milo 0.2.1 and using Kepware server . I get Bad_SessionId exception.