eclipse-milo/milo #294: Error creating connection on IBM JDK+BIG_ENDIAN System

AlthausOventrop, 2018-07-06:
Hello!

I create a small test client app from examples connecting not using security. The same JARs are working on a linux system, but not on the productive System having IBM JDK 1.8 on BIG_ENDIAN.

The error source is (complete output attached in file)
io.netty.handler.codec.DecoderException: UaException: status=Bad_TcpMessageTypeInvalid, message=unknown message type: 4543058 

where HEX(4543058 ) = 455252 which reads 'ERR'

So i am conclusing that there is something misinterpreting the byte-stream in the communication.

Maybe someone can confirm or deny that milo is generally working on IBM JDK / BIG_ENDIAN 
and i'm doing something wrong ?

Any hints appreciated.

Greetings,
Stephan

[miloerr.txt](https://github.com/eclipse/milo/files/2169967/miloerr.txt)

kevinherron, 2018-07-06:
I'm a little stumped by this. My initial suspicion is that it might actually be a bug in Netty... but I'm not sure.

It might be worth trying building and testing your example against the `dev/0.3` branch, which uses a newer version of Netty, which had some endian-related changes.

AlthausOventrop, 2018-07-06:
One more test: 
+ If i use a server on the same big-endian system, this error does not occur.
- if i try a connection from a little-endian client to a big-endian server i get msg type HEY(5391698)=524552 -> 'RER'

So this is definately a problem in the communication between BIG- and LITTLE-endian systems.

AlthausOventrop, 2018-07-06:
"It might be worth trying building and testing your example against the dev/0.3 branch"

as i am new to maven and fellows, 
how do i do this please?

kevinherron, 2018-07-06:
Check out the `dev/0.3` branch and then run `mvn clean install` at the project root. You should then be able to update the Milo dependencies in your examples to `0.3.0-SNAPSHOT` and rebuild them.

Another thing that would be interesting is running the server on the big-endian system and then connecting with another UA client (like UaExpert) from a little-endian system.

kevinherron, 2018-07-06:
If you can capture Wireshark during this process that would be even better. Would be nice to see if the bytes wrong on the wire or if it's happening somewhere in Milo/Netty.

AlthausOventrop, 2018-07-06:
Many thanks for the hints, i will have a try!

AlthausOventrop, 2018-07-06:
Success!
With the dev/0.3 branch i don't get an error and i can read the values as expected.

Now we have the possibility to connect our machines to our database. As i'm out of business for two weeks, i will be able do provide more checks to find the origin of the problem (netty or milo or whatelse) by the end of the month..

Many thanks!!

kevinherron, 2018-07-06:
Okay, glad it's working. I suspect Netty 4.1 fixed it.

closed 2018-07-06 by kevinherron