eclipse-milo/milo #316: ERROR - Call error: Session Call error BadInternalError.

MeowStone, 2018-08-31:


MeowStone, 2018-08-31:
can't use DataValue as a AnnotationBasedInvocationHandler.Out value?

MeowStone, 2018-08-31:
error occured in AnnotationBasedInvocationHandler.java 
-----------------------------------------------------
 for(int i = 0; i < outputCount; ++i) {
        values[i] = new Variant(((AnnotationBasedInvocationHandler.OutImpl)outputs[i]).get());
}
-------------------------------------------------------
exception had been caught here

MeowStone, 2018-08-31:
`checkArgument(!DataValue.class.equals(componentClazz), "Variant cannot contain DataValue");`
how can i return complicate data structure like DataValue?

kevinherron, 2018-08-31:
From the spec:
>DataValue and DiagnosticInfo types only have meaning when returned in a response message
with an associated StatusCode. As a result, Variants cannot contain instances of DataValue or
DiagnosticInfo.

Use a different datatype here. You're misusing `DataValue`.

kevinherron, 2018-09-03:
Well... for the results of method calls you can't.

MeowStone, 2018-09-04:
@kevinherron thks to telling me this, multiple return/out parameters . but, i want to return parameters  like map

MeowStone, 2018-09-04:
okay,so i can't return customtype data right now.
thks anyway :D

kevinherron, 2018-09-04:
Ha, okay. `MethodInvocationHandler` doesn't have to be complicated - all it needs to do is validate the inputs received before invoking the logic.

closed 2018-09-04 by kevinherron