eclipse-milo/milo #322: Timeout Connecting to OPC-UA Server

wsuetholz, 2018-09-14:
Really not a bug report as such.. 

I am currently experiencing the following problem:

`
Exception Caught Connecting: "UaException: status=Bad_Timeout, message=request timed out after 5000ms"
`

More a plea for information on what I need to change to set this to 60000ms instead? In an attempt to do this I've recently added the "timeout = new Long(60000);" and ".setAcknowlegeTimeout(uint(timeout))" lines. With wireshark I see the connection and the authentication negotiations happening, but the timeout occurs while that negotiation is going on. Using the UAExpert client I've clocked the connect time from a low of 8 seconds to higher than the 10 seconds that UAExpert allows for. I've inquired with the vendor in order to speed up their side, but not getting much hope that they will be able to speed their end up enough. The code used to start the connection is below. The exception at the end of the startup method is what is seen in my logs.

I couldn't seem to get the formatting below to be all code...  Not sure why?

`
public void startup() {
    URI uri = null;
    try {
        uri = new URI (url);
    } catch (Exception ex) {
        logger.error("Parse URL to URI Exception:" + ex.getLocalizedMessage());
    }
    String hostname = "localhost";
    int portnum = 56000;

    if (uri != null) {
        hostname = uri.getHost();
        portnum = uri.getPort();
    }

    CompletableFuture<UaClient> request = null;
    logger.info("Startup PbOpcUaClient \"" + url + "\"");

    timeout=new Long(60000);

    try {
// EndpointDescription[] endpoints = UaTcpStackClient.getEndpoints("opc.tcp://localhost:12686/test-server").get();
EndpointDescription[] endpoints = UaTcpStackClient.getEndpoints(url).get();
EndpointDescription endpoint = Arrays.stream(endpoints)
.filter(e -> e.getSecurityPolicyUri().equals(SecurityPolicy.None.getSecurityPolicyUri()))
.findFirst().orElseThrow(() -> new Exception("no desired endpoints returned"));

        endpoint = updateEndpointUrl(endpoint, hostname);

        KeyStoreLoader loader = new KeyStoreLoader().load();

        OpcUaClientConfig clientConfig = OpcUaClientConfig.builder()
                .setApplicationName(LocalizedText.english(appName))
                .setApplicationUri(OPC_URI_LEADIN + appUri)
                .setCertificate(loader.getClientCertificate())
                .setKeyPair(loader.getClientKeyPair())
                .setEndpoint(endpoint)
                .setRequestTimeout(uint(timeout))
                .setAcknowledgeTimeout(uint(timeout))
                .setSessionTimeout(uint(timeout))
                .setIdentityProvider(new UsernameProvider(username, password))
                .build();

        opcUaClient = new OpcUaClient(clientConfig);
        opcUaClient.addFaultListener(this);
        request = opcUaClient.connect();
        uaClient = request.get(60L, TimeUnit.SECONDS);
        if (uaClient == null) {
            connected = false;
        } else {
            connected = true;
            uaSession = uaClient.getSession().get();
        }

        //client.browse();
        logger.info("Client is now Connected.");
    } catch (Exception ex) {
        logger.error("Exception Caught Connecting: \"" + ex.getLocalizedMessage() + "\"");
    }
}
`

kevinherron, 2018-09-14:
My guess is you're running into one of the following hardcoded timeouts:
* https://github.com/eclipse/milo/blob/1660d2f8ed445300f53479e12feb9c79f673322b/opc-ua-stack/stack-client/src/main/java/org/eclipse/milo/opcua/stack/client/UaTcpStackClient.java#L494
* https://github.com/eclipse/milo/blob/1660d2f8ed445300f53479e12feb9c79f673322b/opc-ua-stack/stack-client/src/main/java/org/eclipse/milo/opcua/stack/client/UaTcpStackClient.java#L565

Try changing one or both of those and let me know. These values won't be hardcoded when 0.3.0 is released.

wsuetholz, 2018-09-14:
Thanks for the quick reply.  Do you mean that I have to change it at that level?  I'm not presently compiling your code, I'm using the maven repo.. Well, at least you pointed me directly at what needs to be changed, Thank you.

kevinherron, 2018-09-14:
Yeah, you should be able to clone this repo, make the changes, and run `mvn install` from the project root (make sure you're using JDK8).

Then you can change the version your code depends on to `0.2.4-SNAPSHOT` and it will resolve from your local m2 repo.

kevinherron, 2018-09-14:
There's another hardcoded timeout here:
https://github.com/eclipse/milo/blob/1660d2f8ed445300f53479e12feb9c79f673322b/opc-ua-stack/stack-client/src/main/java/org/eclipse/milo/opcua/stack/client/handlers/UaTcpClientMessageHandler.java#L81

wsuetholz, 2018-09-16:
Thank you for the pointers Kevin.

closed 2018-09-17 by kevinherron