eclipse-milo/milo #326: Java 7 for android platform

vikram919, 2018-10-02:
Does eclipse milo support android platforms especially with jdk 7?

kevinherron, 2018-10-02:
Milo requires Java 8. 

I don't know if it works on any version of Android or not - I no longer have an Android phone I could even test with.

vikram919, 2018-10-02:
Hello @kevinherron ,
Thanks for the message, today I have tested Milo on Android Things (which is basically a Raspberrypi with Android OS) and found a method Not found exception on BigInteger.longValueExact() while Subscribing to server. After further investigation I found [here](https://stackoverflow.com/questions/42550985/android-studio-can-not-find-java-math-biginteger-some-methods-in-project) some relative discussion. Now I assume android does not support all java 1.8 features.

kevinherron, 2018-10-02:
Right, Android supports some subset of the Java 8 API (starting at some SDK version, I don't know which). It's possible to refactor this particular usage to use something other than `longValueExact()` but there's likely other problems that will pop up after that.

closed 2018-10-03 by kevinherron