eclipse-milo/milo #327: Multiple Subscriptions with same Client Handler will Overwrite each other

lelmarir, 2018-10-02:
**Scenario**:
> MonitoringParameters parameters = new MonitoringParameters(...);
> 
> MonitoredItemCreateRequest tag1Request = new MonitoredItemCreateRequest(tag1, MonitoringMode.Reporting, parameters);
> MonitoredItemCreateRequest tag2Request = new MonitoredItemCreateRequest(tag2, MonitoringMode.Reporting, parameters); //same parameters
> 
> subscription.createMonitoredItems(TimestampsToReturn.Both, newArrayList(tag1Request), new CreationCallback()).get();
> subscription.createMonitoredItems(TimestampsToReturn.Both, newArrayList(tag2Request), new CreationCallback()).get();

**Problem**:
While creating two MonitoredItems using the same MonitoringParameters, the second one will internally overwrite the first one.
The resulting notification will report a tag1 value change as if it was a tag2 change; the notified tag value is correct, though.

**Solution**:
Adding two MonitoredItems using the same parameters (same ClientHandle) is wrong (and easy to overlook), but the library does not complain about it. 

It would be enough to throw an exception if a client handle is already used when adding a new item in the map itemsByClientHandle.
https://github.com/eclipse/milo/blob/b4904569cb7bf508d358fa2df665703511fb7246/opc-ua-sdk/sdk-client/src/main/java/org/eclipse/milo/opcua/sdk/client/subscriptions/OpcUaSubscription.java#L130

kevinherron, 2018-10-02:
I don't think throwing an Exception at that point in the code is the right thing to do, but maybe a log message warning about it would be appropriate.

lelmarir, 2018-10-02:
> I don't think throwing an Exception at that point in the code is the right thing to do, but maybe a log message warning about it would be appropriate.

Why? Such condition is not valid and the program should not continue, however even a log would help

kevinherron, 2018-10-02:
Such is a condition *is* explicitly valid, event it's usually an error by the user, but it's intentional that you can assign or re-assign handles.

Further, if you only had one duplicate among multiple successfully created items, throwing an Exception at that point in the code would cause a failure for all of them rather than indicating which item(s) used duplicate handles.

I'd like to do something to help prevent this problem but I just don't think throwing an Exception at that point is the right something.

closed 2018-10-28 by kevinherron