eclipse-milo/milo #330: How to execute opc ua client ?

ericccccccc, 2018-10-03:
Thanks for your suggestions using  the  IDE to build the project.
Now i can execute the server,but i don't how to exeute the client in the eclipse IDE.
I have found there are  a lot of client example.
I want to execute the basic ExampleClient but there isn't have the main .
how to execute ?

thank you !!

kevinherron, 2018-10-03:
There's no basic ExampleClient, just specific examples like `ReadExample` that *do* have a `main` you can start.

You don't need to start a server for these, they start one for you.

ericccccccc, 2018-10-04:
thx~for your comment.
i have tried to start the ReadExample :
/client-examples$ mvn exec:java -D"exec.mainClass"="org.eclipse.milo.examples.client.ReadExample"

but i got the 2 error ~
[image]

[image]

and you say that "You don't need to start a server for these, they start one for you." , is that means i only need to start ReadExample then the ExampleServer will start too ?

I still got a lot problems, but thx for your help ^^

kevinherron, 2018-10-04:
> and you say that "You don't need to start a server for these, they start one for you." , is that means i only need to start ReadExample then the ExampleServer will start too ?

Yes, that's why you get the BindException in your first screenshot.

ericccccccc, 2018-10-05:
thx  a lot!!

closed 2018-10-05 by kevinherron