eclipse-milo/milo #331: Problems encountered in reading historical data

tiedun001, 2018-10-08:
The following problems were encountered when reading the historical data. 
[HistoryReadResult{StatusCode=StatusCode{name=Bad_NotSupported, value=0x803D0000, quality=bad}, ContinuationPoint=null, HistoryData=null}]
the point is bad_notsupported what's the problem?
please help me!

kevinherron, 2018-10-08:
The server you're connecting to probably doesn't support historical services.

kevinherron, 2018-10-09:
If you're using Milo SDK for the server, you've got quite a bit of work ahead of you. An implementation of historical services is not part of the SDK.

You can implement them yourself by overriding `AttributeHistoryServices#historyRead` and `AttributeHistoryServices#historyUpdate` in your `Namespace` implementation.

You'll need Part 4 and Part 11 of the OPC UA spec, at a minimum.

tiedun001, 2018-10-09:
thank you.i  use Milo SDK for ua server.  I want to know the details of implementation.

kevinherron, 2018-10-09:
The details of how to implement historical services are in Part 4 and Part 11 of the OPC UA specification.

I haven't implemented it myself so I can't provide any further help.

tiedun001, 2018-10-09:
or other sdk has finished ?

kevinherron, 2018-10-09:
> or other sdk has finished ?
I don't know.

There's no other open source Java SDK. One of the commercial SDKs may have history implemented for you to some degree.

closed 2018-10-09 by kevinherron