eclipse-milo/milo #332: Uncaught Exception when OPC UA connection terminates unexpectedly

danyadibence, 2018-10-08:
Hi,

I would like to test what happens if the OPC UA client (developed using Eclipse Milo) loses connection to a server (ProSys OPC UA Simulation Server), because the server is not available any more. I tested the connection's termination by closing the simulation server.
However, in this case the following exception will be thrown and logged regularly:
```
2018-10-08 14:45:24.751 ERROR 19244 --- [global-threads)] c.m.m.c.l.o.OPCConnectionManager         : Failed to close OPC ua connection to the machine with the id 1, displayed name Test, displayed number 1.

java.util.concurrent.TimeoutException: null
	at java.util.concurrent.CompletableFuture.timedGet(CompletableFuture.java:1886) ~[?:?]
	at java.util.concurrent.CompletableFuture.get(CompletableFuture.java:2021) ~[?:?]
	at com.mmmgroup.module.cssd.livedata.opc.OPCConnection.synchronousDisconnect(OPCConnection.java:240) ~[main/:?]
	at com.mmmgroup.module.cssd.livedata.opc.OPCConnectionManager.closeConnection(OPCConnectionManager.java:210) ~[main/:?]
	at java.util.concurrent.ConcurrentHashMap$KeySetView.forEach(ConcurrentHashMap.java:4724) [?:?]
	at com.mmmgroup.module.cssd.livedata.opc.OPCConnectionManager.closeAllConnections(OPCConnectionManager.java:197) [main/:?]
	at com.mmmgroup.module.cssd.livedata.messaging.LiveDataModuleShutDownFrameAgentListener.onFrame(LiveDataModuleShutDownFrameAgentListener.java:69) [main/:?]
	at com.mmmgroup.core.messaging.base.BaseFrameAgent.handOver(BaseFrameAgent.java:155) [mmm-messaging-0.3.3-RC2.jar:0.3.3-RC2]
	at com.mmmgroup.core.messaging.base.BaseFrameAgent.handleFrameInternal(BaseFrameAgent.java:130) [mmm-messaging-0.3.3-RC2.jar:0.3.3-RC2]
	at com.mmmgroup.core.messaging.base.AbstractFrameHandler.handleFrame(AbstractFrameHandler.java:81) [mmm-messaging-0.3.3-RC2.jar:0.3.3-RC2]
	at com.mmmgroup.core.messaging.artemis.ArtemisMessageHandler.onMessage(ArtemisMessageHandler.java:81) [mmm-messaging-apache-artemis-0.2.10-RELEASE.jar:0.2.10-RELEASE]
	at org.apache.activemq.artemis.core.client.impl.ClientConsumerImpl.callOnMessage(ClientConsumerImpl.java:1003) [artemis-core-client-2.4.0.jar:2.4.0]
	at org.apache.activemq.artemis.core.client.impl.ClientConsumerImpl.access$400(ClientConsumerImpl.java:50) [artemis-core-client-2.4.0.jar:2.4.0]
	at org.apache.activemq.artemis.core.client.impl.ClientConsumerImpl$Runner.run(ClientConsumerImpl.java:1126) [artemis-core-client-2.4.0.jar:2.4.0]
	at org.apache.activemq.artemis.utils.actors.OrderedExecutor.doTask(OrderedExecutor.java:42) [artemis-commons-2.4.0.jar:2.4.0]
	at org.apache.activemq.artemis.utils.actors.OrderedExecutor.doTask(OrderedExecutor.java:31) [artemis-commons-2.4.0.jar:2.4.0]
	at org.apache.activemq.artemis.utils.actors.ProcessorBase$ExecutorTask.run(ProcessorBase.java:53) [artemis-commons-2.4.0.jar:2.4.0]
	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1167) [?:?]
	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:641) [?:?]
	at java.lang.Thread.run(Thread.java:844) [?:?]
```

Is there a way to handle this exception so that it won't appear in the logs all the time? It might cause the log files to contain unnecessary informations about the lost connection.

Thank you very much in advance!

danyadibence, 2018-10-08:
Yes, it would translate to something similar. According to [this](https://github.com/netty/netty/issues/4897) issue the precise description is "An established connection was aborted by the software in your host machine"

kevinherron, 2018-10-08:
What version of Milo are you using? I'm not able to reproduce this.

danyadibence, 2018-10-09:
I'm using 0.2.3.
I forgot to add in my previous comments that I have variable subscriptions, sorry for that.

kevinherron, 2018-10-09:
I still can't reproduce this.

The Exception you're getting is suspicious - it's one I'd expect to see if e.g. the network adapter was disabled or your IP address changed or something, not one you should see from just closing the simulation server. Can you tell me more about how you're testing this? OS? Client and server on same machine?

I think this is just a matter of override `exceptionCaught` in one of the handlers (maybe `UaTcpClientAcknoledgeHandler`, but I can't be sure without reproducing it. If you're in a position to make some changes and recompile them yourself that would help too.

danyadibence, 2018-10-09:
Thank you for your help. The server and client are on the same Windows 10 machine.

I'll try overriding an exceptionCaught() method.

danyadibence, 2018-10-10:
Hi @kevinherron,

I managed to solve the problem, it was caused by a thread used for a different use-case.
I reorganised my threads and their synchronization and now the redundant log messages aren't visible any more.

kevinherron, 2018-10-10:
Ok, great :+1:

closed 2018-10-10 by kevinherron

Po252, 2022-03-24:
> Hi @kevinherron,
> 
> I managed to solve the problem, it was caused by a thread used for a different use-case. I reorganised my threads and their synchronization and now the redundant log messages aren't visible any more.
Hi @danyadibence 
help😂,my milo version is 0.2.4 .But I don't know this threads should be reorganized

tsmortenson, 2025-01-21:
We also still see this regularly while shutting down out application with version 0.6.15

```
ERROR  *[k.client.transport.uasc.UascClientMessageHandler] (milo-netty-event-loop-5                                              ): [remote=kwi-lacs-y7-1/10.210.1.10:52240] Exception caught: 既存の接続はリモート ホストに強制的に切断されました。
java.io.IOException: 既存の接続はリモート ホストに強制的に切断されました。
      at sun.nio.ch.SocketDispatcher.read0(Native Method) ~[?:?]
      at sun.nio.ch.SocketDispatcher.read(SocketDispatcher.java:43) ~[?:?]
      at sun.nio.ch.IOUtil.readIntoNativeBuffer(IOUtil.java:276) ~[?:?]
      at sun.nio.ch.IOUtil.read(IOUtil.java:233) ~[?:?]
      at sun.nio.ch.IOUtil.read(IOUtil.java:223) ~[?:?]
      at sun.nio.ch.SocketChannelImpl.read(SocketChannelImpl.java:356) ~[?:?]
      at io.netty.buffer.PooledByteBuf.setBytes(PooledByteBuf.java:255) ~[netty-buffer-4.1.112.Final.jar:4.1.112.Final]
      at io.netty.buffer.AbstractByteBuf.writeBytes(AbstractByteBuf.java:1132) ~[netty-buffer-4.1.112.Final.jar:4.1.112.Final]
      at io.netty.channel.socket.nio.NioSocketChannel.doReadBytes(NioSocketChannel.java:356) ~[netty-transport-4.1.112.Final.jar:4.1.112.Final]
      at io.netty.channel.nio.AbstractNioByteChannel$NioByteUnsafe.read(AbstractNioByteChannel.java:151) ~[netty-transport-4.1.112.Final.jar:4.1.112.Final]
      at io.netty.channel.nio.NioEventLoop.processSelectedKey(NioEventLoop.java:788) ~[netty-transport-4.1.112.Final.jar:4.1.112.Final]
      at io.netty.channel.nio.NioEventLoop.processSelectedKeysOptimized(NioEventLoop.java:724) ~[netty-transport-4.1.112.Final.jar:4.1.112.Final]
      at io.netty.channel.nio.NioEventLoop.processSelectedKeys(NioEventLoop.java:650) ~[netty-transport-4.1.112.Final.jar:4.1.112.Final]
      at io.netty.channel.nio.NioEventLoop.run(NioEventLoop.java:562) ~[netty-transport-4.1.112.Final.jar:4.1.112.Final]
      at io.netty.util.concurrent.SingleThreadEventExecutor$4.run(SingleThreadEventExecutor.java:994) ~[netty-common-4.1.112.Final.jar:4.1.112.Final]
      at io.netty.util.internal.ThreadExecutorMap$2.run(ThreadExecutorMap.java:74) ~[netty-common-4.1.112.Final.jar:4.1.112.Final]
      at java.lang.Thread.run(Thread.java:829) ~[?:?]
```

We ended up adding a lot of allocate/release code to make sure the OpcUaClient and its NodeManager are never disposed while they are still in use.   Then also made sure toe delete any subscriptions that had been made.
What finally fixed the problem though was to change the code:
```
opcUaClient.disconnect();
opcUaClient = null;
```
to:
```
opcUaClient.disconnect().get();
opcUaClient = null;
```
This seems to make sure the code waits for the disconnect to actually complete before proceeding to dispose the OpcUaClient instance during shutdown of the JVM.

Hope this helps.