eclipse-milo/milo #352: LDS feature in Milo

blasty3, 2018-11-01:
Hello @kevinherron / @Pro ,

I noticed that there is a separate branch for the LDS feature https://github.com/eclipse/milo/tree/feature/lds (apparently is not yet merged to the master branch)? Just wondering if the LDS branch is somehow usable? I'm planning to use it for learning & research purpose. 

Best,

kevinherron, 2018-11-01:
It might be usable if you check it out and run it as-is, but it's long past the point where it can be merged successfully.

I'm keeping it around because eventually I'll re-implement the functionality and I want to re-use some of the mDNS logic that was written there.

kevinherron, 2018-11-06:
It's not a mistake, this isn't a copy of the foundation code. Different design decisions were made.

blasty3, 2018-11-06:
Hello,

Can you elaborate a little bit regarding these design decisions you are referring to?
Also, just wondering if you wouldn't mind commenting or enlightening me about my previous post. 

Best,

kevinherron, 2018-11-06:
> Can you elaborate a little bit regarding these design decisions you are referring to?

Either the server needs to know how to register itself or some external entity needs to know how to register the server, and the helper route means that you don't have to couple the server to the knowledge of how to do registration.

I don't know the answers to your previous questions. The work in that branch is a years old and nobody has looked at it since then. It doesn't compile any more and would need a lot of work to make it compile again, starting with a rebase against `dev/0.3` or just starting from scratching and pulling in parts that worked.

closed 2018-12-05 by kevinherron