eclipse-milo/milo #355: 0.3 NodeFactory ModellingRule

Molokay, 2018-11-08:
Hello,
i am using the 0.3 snapshot and have found a new issue with the node factory.
Nodes created by the node factory miss the HasModellingRule references.
The screenshots taken from UAExpert illustrate the problem.
HydraulicPressDeviceType uses the node factory to create an instance of slide unit type named "test". 
Attached is my current project if needed for replication.

[emudig.zip](https://github.com/eclipse/milo/files/2562362/emudig.zip)

[image]
[image]

kevinherron, 2018-11-08:
Instances aren't supposed to have modeling rule references, they only belong on the type definitions.

Molokay, 2018-11-08:
In this case i guess i need to add them manually.
Just to state the problem here if someone else is in the same situation:
In my example the instance of SlideUnitType "test" is part of another type definition, the HydraulicPressDeviceType. Now when i use the node factory to create instances of HydraulicPressDeviceType, due to the missing HasModellingRule references the instances of HydraulicPressDevice miss components and properties of "test".

kevinherron, 2018-11-08:
Yeah, you'll have to add them manually if you're trying to add `test` as an additional instance declaration to the `HydraulicPressType`. 

This kind of thing would ideally be done in a modeling tool and you wouldn't need to assemble all these types and references by hand.

Molokay, 2018-11-08:
Ideally yes. Unless i am mistaken there is no support for importing xml nodesets yet ?
Thank you anyway for the quick response !