eclipse-milo/milo #372: Problem with multible method calls of DelegatingAttributeDelegate

elilueb, 2018-11-27:
Hi
I wondered why each method of the class org.eclipse.milo.opcua.sdk.server.nodes.delegates.DelegatingAttributeDelegate is called twice after clicking on a file or variable via an OPC-UA client. My main problem is not that the methods are called twice, but the firs time getAccessLevel() is called, the context attribute doesn’t has a session. For me this is quite complicated to handle.

kevinherron, 2018-11-27:
It depends what that UA client is doing when you click on the node. When it's called without a session that means it's an "internal" read, i.e. the server needs the value of that attribute in order to fulfill some other service call.

kevinherron, 2018-12-05:
Did you ever find out what calls were happening?

closed 2019-02-18 by kevinherron