eclipse-milo/milo #388: Instantiation of ObjectType with layered components

Eichi87, 2019-01-08:
As mentioned in the title, we try to instantiate an objecttype containing a component which contains another component.

The screenshots below show that this doesn't seem to work.

The Object:
[image]

The Node without the second component:
[image]

The code we use to instantiate the object is this:
```
this.uaObjectNode = new UaObjectNode[xmlImporter.getNumberOfTotalObjectTypes()];

        for(int i=0; i<this.uaObjectNode.length ; i++) {

           this.uaObjectNode[i] = nodeFactory.createObject(

                    new NodeId(2, "11000"),

                    new QualifiedName(2, this.xmlObjectType[i].getBrowseName()),

                    LocalizedText.english(this.xmlObjectType[i].getDisplayName()),

                    this.uaObjectTypeNode[i].getNodeId()

                );

           this.folderNode.addOrganizes(this.uaObjectNode[i]);
```

Should this work? Or does it work as intended?
Or are we missing something?

Thanks in advance
Daniel

kevinherron, 2019-01-08:
Can you try this on the `dev/0.3` branch instead? NodeFactory was rewritten to fix these issues (hopefully).

Eichi87, 2019-01-14:
The issue is resolved. Our mistake was, that we used types instead of components during the type definition.

kevinherron, 2019-01-14:
Okay 👍

closed 2019-01-14 by kevinherron