eclipse-milo/milo #392: java.util.concurrent.TimeoutException: null

zwjsoft, 2019-01-21:
In Subscription Example, errors occur after receiving events about 10 times。

10:31:31.245 [main] ERROR o.e.m.e.client.ClientExampleRunner - Error running client example: null
java.util.concurrent.TimeoutException: null
	at java.util.concurrent.CompletableFuture.timedGet(CompletableFuture.java:1771)
	at java.util.concurrent.CompletableFuture.get(CompletableFuture.java:1915)
	at org.eclipse.milo.examples.client.ClientExampleRunner.run(ClientExampleRunner.java:145)
	at org.eclipse.milo.examples.client.SubscriptionExample.main(SubscriptionExample.java:46)

zwjsoft, 2019-01-22:
I've lifted the five-minute limit.I use Prosys OPC UA Simulation Sever.
The code is as follows:

public class SubscriptionExample implements ClientExample {

    public static void main(String[] args) throws Exception {
        SubscriptionExample example = new SubscriptionExample();

        new ClientExampleRunner(example).run();
    }

    private final Logger logger = LoggerFactory.getLogger(getClass());

    private final AtomicLong clientHandles = new AtomicLong(1L);

    @Override
    public void run(OpcUaClient client, CompletableFuture<OpcUaClient> future) throws Exception {
        // synchronous connect
        client.connect().get();

        // create a subscription @ 1000ms
        UaSubscription subscription = client.getSubscriptionManager().createSubscription(10000.0).get();

        // subscribe to the Value attribute of the server's CurrentTime node
        ReadValueId readValueId = new ReadValueId(
            //Identifiers.Server_ServerStatus_CurrentTime,
            new NodeId(5, "Counter1"),
            AttributeId.Value.uid(), null, QualifiedName.NULL_VALUE);

        // important: client handle must be unique per item
        UInteger clientHandle = uint(clientHandles.getAndIncrement());

        MonitoringParameters parameters = new MonitoringParameters(
            clientHandle,
            1000.0,     // sampling interval
            null,       // filter, null means use default
            uint(10),   // queue size
            true        // discard oldest
        );

        MonitoredItemCreateRequest request = new MonitoredItemCreateRequest(
            readValueId, MonitoringMode.Reporting, parameters);

        // when creating items in MonitoringMode.Reporting this callback is where each item needs to have its
        // value/event consumer hooked up. The alternative is to create the item in sampling mode, hook up the
        // consumer after the creation call completes, and then change the mode for all items to reporting.
        BiConsumer<UaMonitoredItem, Integer> onItemCreated =
            (item, id) -> item.setValueConsumer(this::onSubscriptionValue);

        List<UaMonitoredItem> items = subscription.createMonitoredItems(
            TimestampsToReturn.Both,
            newArrayList(request),
            onItemCreated
        ).get();

        for (UaMonitoredItem item : items) {
            if (item.getStatusCode().isGood()) {
                logger.info("item created for nodeId={}", item.getReadValueId().getNodeId());
            } else {
                logger.warn(
                    "failed to create item for nodeId={} (status={})",
                    item.getReadValueId().getNodeId(), item.getStatusCode());
            }
        }

        // let the example run for 5 seconds then terminate
        //Thread.sleep(50000);
        //future.complete(client);
    }

[... 154 lines cut ...]
11:37:46.496 [main] INFO  o.e.m.opcua.sdk.client.OpcUaClient - Eclipse Milo OPC UA Client SDK version: 0.2.5-SNAPSHOT
11:37:56.097 [NonceUtilSecureRandom] INFO  o.e.m.o.stack.core.util.NonceUtil - SecureRandom seeded in 0ms.
11:37:57.742 [main] INFO  o.e.m.e.client.SubscriptionExample - item created for nodeId=NodeId{ns=5, id=Counter1}
11:38:07.472 [ua-shared-pool-2] INFO  o.e.m.e.client.SubscriptionExample - subscription value received: item=NodeId{ns=5, id=Counter1}, value=Variant{value=36}
11:38:07.473 [ua-shared-pool-2] INFO  o.e.m.e.client.SubscriptionExample - subscription value received: item=NodeId{ns=5, id=Counter1}, value=Variant{value=37}
11:38:07.473 [ua-shared-pool-2] INFO  o.e.m.e.client.SubscriptionExample - subscription value received: item=NodeId{ns=5, id=Counter1}, value=Variant{value=38}
11:38:07.473 [ua-shared-pool-2] INFO  o.e.m.e.client.SubscriptionExample - subscription value received: item=NodeId{ns=5, id=Counter1}, value=Variant{value=39}
11:38:07.473 [ua-shared-pool-2] INFO  o.e.m.e.client.SubscriptionExample - subscription value received: item=NodeId{ns=5, id=Counter1}, value=Variant{value=40}
11:38:07.474 [ua-shared-pool-2] INFO  o.e.m.e.client.SubscriptionExample - subscription value received: item=NodeId{ns=5, id=Counter1}, value=Variant{value=41}
11:38:07.474 [ua-shared-pool-2] INFO  o.e.m.e.client.SubscriptionExample - subscription value received: item=NodeId{ns=5, id=Counter1}, value=Variant{value=42}
11:38:07.474 [ua-shared-pool-2] INFO  o.e.m.e.client.SubscriptionExample - subscription value received: item=NodeId{ns=5, id=Counter1}, value=Variant{value=43}
11:38:07.474 [ua-shared-pool-2] INFO  o.e.m.e.client.SubscriptionExample - subscription value received: item=NodeId{ns=5, id=Counter1}, value=Variant{value=44}
11:38:07.474 [ua-shared-pool-2] INFO  o.e.m.e.client.SubscriptionExample - subscription value received: item=NodeId{ns=5, id=Counter1}, value=Variant{value=45}
11:38:12.809 [main] ERROR o.e.m.e.client.ClientExampleRunner - Error running client example: null
java.util.concurrent.TimeoutException: null
	at java.util.concurrent.CompletableFuture.timedGet(CompletableFuture.java:1771)
	at java.util.concurrent.CompletableFuture.get(CompletableFuture.java:1915)
	at org.eclipse.milo.examples.client.ClientExampleRunner.run(ClientExampleRunner.java:145)
	at org.eclipse.milo.examples.client.SubscriptionExample.main(SubscriptionExample.java:46)
11:38:12.810 [main] ERROR o.e.m.e.client.ClientExampleRunner - Error running example: null
java.util.concurrent.TimeoutException: null
	at java.util.concurrent.CompletableFuture.timedGet(CompletableFuture.java:1771)
	at java.util.concurrent.CompletableFuture.get(CompletableFuture.java:1915)
	at org.eclipse.milo.examples.client.ClientExampleRunner.run(ClientExampleRunner.java:145)
	at org.eclipse.milo.examples.client.SubscriptionExample.main(SubscriptionExample.java:46)

kevinherron, 2019-01-22:
`ClientExampleRunner` expects examples to complete within a reasonable time before that timeout occurs. All you've done is guarantee the subscription example no longer completes successfully.

https://github.com/eclipse/milo/blob/5e9242c0b62260ac34963bc75a7765b12771c4f8/milo-examples/client-examples/src/main/java/org/eclipse/milo/examples/client/ClientExampleRunner.java#L143-L149