eclipse-milo/milo #398: Instance of nested object does not produce object variables

CB-IT, 2019-02-03:
Hi,
in relation to the provided example namespace I modeled 3 ObjectTypes:
1. A simple MyObjectType1 with two variables
--> Works fine, when i instance UaObjectNode with server.getNodeFactory().createNode(...)

2. MyObjectType2 with a variable and an object of type MyObjectType1 
I add the object as UaObjectNode with server.getNodeFactory().createNode(...)
--> In ObjectTypes view it looks good
--> But the instance of MyObjectType2 only contains an empty object. The variables of "value" are missing.
--> Furthermore, the ModellingRule-References from MyObjectType1 are not copied to the variables of "value" 

3. MyObjectType3 extending MyObjectType1 
--> Works fine, when i instance UaObjectNode with server.getNodeFactory().createNode(...)

I uploaded the code: [gist](https://gist.github.com/CB-IT/b96fa6f61f53063c82436de1b5e7b9ab)

Here you can see the result in UAExpert:
[image]

Is it my fault or a currently unsupported feature?

kevinherron, 2019-02-03:
I believe this issue is fixed with the new `NodeFactory` implementation on the `dev/0.3` branch. 

If you're already using that branch and it's not working please let me know and I'll investigate further.

CB-IT, 2019-02-03:
Sorry, but i am on dev/0.3. I did a pull without changes and a test before open this issue.

kevinherron, 2019-02-03:
Ok thanks, I'll take a closer look at your example and see what I find.

CB-IT, 2019-02-04:
Ok, thanks for the hint. It's working know.
But I thought, that all is taken from the type definition. In this way we have to model things twice.

kevinherron, 2019-02-04:
The verbosity is unfortunate but AFAIK this is all required by the spec.

If you look at the output from code generation tools you'll see the modeling rules are present on all the nodes, recursively, in the hierarchy of any given type.

closed 2019-02-04 by kevinherron