eclipse-milo/milo #408: Manually setting endpointURL for connection

ElreboCM, 2019-02-21:
Maybe I am missing something but is it somehow possible to force milo to connect to a manually specified endpoint? The example allows uses discovers the endpoints and chooses one from the list.
Setting the endpointUrl in the configuration does not help as only the endpoint description is applied for the connection.

I have the case in which the opc-ua server's endpoint URL are configured to the machine name but I cannot reach the machine without an additional suffix. Unfortunately I cannot change the server configuration.

kevinherron, 2019-02-21:
When you receive `EndpointDescription`s with hostnames you can't resolve you can replace them using  [`EndpointUtil.updateUrl`](https://github.com/eclipse/milo/blob/abab5e38a3148d17a1a3ca9e26843ca267b6c14a/opc-ua-stack/stack-core/src/main/java/org/eclipse/milo/opcua/stack/core/util/EndpointUtil.java#L70).

kevinherron, 2019-02-21:
Errr, not sure what you meant by this:

> Setting the endpointUrl in the configuration does not help as only the endpoint description is applied for the connection.

Now I'm not sure what you're asking.

ElreboCM, 2019-02-21:
There is a method for setting the EndpointUrl in the configuration (**setEndpointUrl**).
Example:

 EndpointDescription[] endpoints = UaTcpStackClient.getEndpoints(_endpoint)
                .get();
        OpcUaClientConfigBuilder cfg = new OpcUaClientConfigBuilder();
        cfg.setEndpoint(endpoints[0]);
        cfg.setEndpointUrl(_endpoint);
        client.connect().get();

However, this does not have an effect to which endpoint the client connects.

kevinherron, 2019-02-21:
You'll need to update the endpoint URL in your EndpointDescription and ignore that setting on the config object.

I think that was intended to be used for the initial discovery before you had an EndointDescription to use. You'll find in the `dev/0.3` branch this has been removed.

ElreboCM, 2019-02-21:
I have tested using EndPointUtil.updateUrl but this also does not have an effect to the URL the client tries to connect. 

`EndpointDescription[] endpoints = UaTcpStackClient.getEndpoints(_endpoint)
                .get();
        OpcUaClientConfigBuilder cfg = new OpcUaClientConfigBuilder();
        cfg.setEndpoint(endpoints[0]);
        EndpointUtil.updateUrl(endpoint,"opc.tcp://10.240.22.1", 48010);
 client  = new OpcUaClient(cfg.build());
        client.connect().get();`

The client will still connect to the original URL.

ElreboCM, 2019-02-21:
How can I ignore the setting in the config object?

kevinherron, 2019-02-21:
The function signature, documentation, and implementation of that function all make it fairly clear that a new EndointDescription object is returned. Try using that.

e.g. 
```java
EndpointDescription updated = EndpointUtil.updateUrl(endpoints[0], "10.240.22.1", 48010);
cfg.setEndpoint(updated);
```

ElreboCM, 2019-02-21:
Sorry I missed that. 🤦

kevinherron, 2019-02-21:
> Sorry I missed that. 🤦

No sweat. You'll find all of the UA structures and anything else related, where possible, are immutable.

closed 2019-02-21 by kevinherron