eclipse-milo/milo #419: BuiltinDataTypeDictionaryInitializerTest.testInitialize test fails.

KairatRakhimov, 2019-03-14:
On Windows 10 machine with Oracle JDK 8, following test fails:

`testInitialize
java.lang.AssertionError: null
	at org.eclipse.milo.opcua.stack.core.types.BuiltinDataTypeDictionaryInitializerTest.testInitialize(BuiltinDataTypeDictionaryInitializerTest.java:37)`

The problem is in Guava code v.19 and described here (https://github.com/google/guava/issues/3402).
However, this issue was allready fixed [here](https://github.com/google/guava/blob/master/guava/src/com/google/common/reflect/ClassPath.java#L586) and released in version **23.6-jre**.

kevinherron, 2019-03-14:
Can you provide the full stack trace for the error? This looks like the same thing that happens when you don't compile with JDK 8.

KairatRakhimov, 2019-03-14:
Yes sure
[milo-test-fail.log](https://github.com/eclipse/milo/files/2966355/milo-test-fail.log)

In  BuiltinDataTypeDictionaryInitializerTest.testInitialize(), existence of org.eclipse.milo.opcua.stack.core.types.structured package classes is checked. 

Guava method ClassPath.from(Classloader) is used to retrieve classpath jar's and folders if any.
Classes are in place, but due to mentioned issue in Guava v.19, used Guava method returns 0 sized collection.

I have tested with Guava v.23.6-jre, it works, tests are passing, but i can say nothing about stability.

kevinherron, 2019-03-14:
Ok, thank you. I'll have to see if upgrading Guava in the 0.2.x branch is feasible. I'm not sure there's even going to be another release in this series.

I've already upgraded Guava on `dev/0.3`, which I'm hoping to release by April. Can you try building there to make sure it works?

KairatRakhimov, 2019-03-15:
Hi Kevin,
Cool :)  I will use dev/0.3 than. I've just tested it, all tests are passing. issue fixed in dev/0.3

closed 2019-03-15 by kevinherron