eclipse-milo/milo #420: Server capabilities: LocaleIdArray

stud0709, 2019-03-14:
Hello Kevin, 

an OPC UA server can support multiple locales (this is why LocaleId is an array). But an UaNode has only one LocalizedText description. 

Is it possible to have multiple locales with milo?

Best Regards, 

Yuriy

kevinherron, 2019-03-14:
It's not well supported right now. 

You'd have to implement it by intercepting at your `Namespace`'s read() implementation or by setting an `AttributeDelegate` on the UaNode that intercepted calls to read the description.

But it looks like the user's locale ids from `ActivateSession` aren't even set on the `Session` right now, so I think some work will need to happen.

kevinherron, 2019-03-14:
tracking in #421

closed 2019-03-14 by kevinherron

kevinherron, 2019-03-19:
@stud0709 this has been implemented