eclipse-milo/milo #422: ConcurrentModificationException by using of NodeFactory in multi threading env.

eugeis, 2019-03-15:
Hello,

I am using dev/0.3 branch.

I am getting ConcurrentModificationException by calling of org.eclipse.milo.opcua.sdk.server.nodes.factories.NodeFactory.createNode(...) in multi threading env.

Please see attached file.

It is ok to use NodeFactory.createNode(...) in multi threading env. or not?

Thanks

eugeis, 2019-03-15:
I don't see how to attached a file, therefore here the output:

`	java.util.ConcurrentModificationException: null,
	at java.base/java.util.ArrayList$Itr.checkForComodification(ArrayList.java:937),
	at java.base/java.util.ArrayList$Itr.next(ArrayList.java:891),
	at com.google.common.collect.AbstractMapBasedMultimap$WrappedCollection$WrappedIterator.next(AbstractMapBasedMultimap.java:466),
	at java.base/java.util.AbstractCollection.addAll(AbstractCollection.java:351),
	at java.base/java.util.LinkedHashSet.<init>(LinkedHashSet.java:169),
	at org.eclipse.milo.opcua.sdk.server.api.AbstractNodeManager.getReferences(AbstractNodeManager.java:95),
	at org.eclipse.milo.opcua.sdk.server.api.DelegatingNodeManager.getReferences(DelegatingNodeManager.java:89),
	at org.eclipse.milo.opcua.sdk.server.nodes.UaNode.getReferences(UaNode.java:201),
	at org.eclipse.milo.opcua.sdk.server.nodes.factories.InstanceDeclarationHierarchy$Builder.hasModellingRule(InstanceDeclarationHierarchy.java:153),
	at org.eclipse.milo.opcua.sdk.server.nodes.factories.InstanceDeclarationHierarchy$Builder.isInstanceDeclaration(InstanceDeclarationHierarchy.java:149),
	at org.eclipse.milo.opcua.sdk.server.nodes.factories.InstanceDeclarationHierarchy$Builder.lambda$null$6(InstanceDeclarationHierarchy.java:126),
	at java.base/java.util.Optional.ifPresent(Optional.java:172),
	at org.eclipse.milo.opcua.sdk.server.nodes.factories.InstanceDeclarationHierarchy$Builder.lambda$addModeledNodes$7(InstanceDeclarationHierarchy.java:119),
	at java.base/java.util.ArrayList.forEach(ArrayList.java:1378),
	at org.eclipse.milo.opcua.sdk.server.nodes.factories.InstanceDeclarationHierarchy$Builder.addModeledNodes(InstanceDeclarationHierarchy.java:118),
	at org.eclipse.milo.opcua.sdk.server.nodes.factories.InstanceDeclarationHierarchy$Builder.lambda$null$6(InstanceDeclarationHierarchy.java:135),
	at java.base/java.util.Optional.ifPresent(Optional.java:172),
	at org.eclipse.milo.opcua.sdk.server.nodes.factories.InstanceDeclarationHierarchy$Builder.lambda$addModeledNodes$7(InstanceDeclarationHierarchy.java:119),
	at java.base/java.util.ArrayList.forEach(ArrayList.java:1378),
	at org.eclipse.milo.opcua.sdk.server.nodes.factories.InstanceDeclarationHierarchy$Builder.addModeledNodes(InstanceDeclarationHierarchy.java:118),
	at org.eclipse.milo.opcua.sdk.server.nodes.factories.InstanceDeclarationHierarchy$Builder.buildHierarchyForType(InstanceDeclarationHierarchy.java:107),
	at org.eclipse.milo.opcua.sdk.server.nodes.factories.InstanceDeclarationHierarchy$Builder.build(InstanceDeclarationHierarchy.java:96),
	at org.eclipse.milo.opcua.sdk.server.nodes.factories.InstanceDeclarationHierarchy.create(InstanceDeclarationHierarchy.java:72),
	at org.eclipse.milo.opcua.sdk.server.nodes.factories.InstanceDeclarationHierarchy$Builder.lambda$build$2(InstanceDeclarationHierarchy.java:94),
	at java.base/java.util.Optional.map(Optional.java:254),
	at org.eclipse.milo.opcua.sdk.server.nodes.factories.InstanceDeclarationHierarchy$Builder.build(InstanceDeclarationHierarchy.java:93),
	at org.eclipse.milo.opcua.sdk.server.nodes.factories.InstanceDeclarationHierarchy.create(InstanceDeclarationHierarchy.java:72),
	at org.eclipse.milo.opcua.sdk.server.nodes.factories.InstanceDeclarationHierarchy$Builder.lambda$build$2(InstanceDeclarationHierarchy.java:94),
	at java.base/java.util.Optional.map(Optional.java:254),
	at org.eclipse.milo.opcua.sdk.server.nodes.factories.InstanceDeclarationHierarchy$Builder.build(InstanceDeclarationHierarchy.java:93),
	at org.eclipse.milo.opcua.sdk.server.nodes.factories.InstanceDeclarationHierarchy.create(InstanceDeclarationHierarchy.java:72),
	at org.eclipse.milo.opcua.sdk.server.nodes.factories.NodeFactory.createNodeTree(NodeFactory.java:101),
	at org.eclipse.milo.opcua.sdk.server.nodes.factories.NodeFactory.createNode(NodeFactory.java:61),`

kevinherron, 2019-03-15:
It *should* be safe, but something in AbstractNodeManager apparently is not.

I'm working on big changes in the area of AddressSpace / NodeManagement that should resolve the issue. For now, you'll have to synchronize access manually.

eugeis, 2019-03-15:
ok, thanks.

kevinherron, 2019-03-15:
Check back in Monday perhaps, I should be done by then. 

There have been some API changes that will require updates to server code.

closed 2019-03-15 by kevinherron

kevinherron, 2019-03-18:
@eugeis I doubt I’ll merge this on Monday, but sometime next week, hopefully after some feedback and more testing.

If you want a head start it’s this branch: 
https://github.com/eclipse/milo/tree/0.3-address-space