eclipse-milo/milo #449: Is Eclipse Milo Officially Supported by OPC UA?

joesan, 2019-04-09:
Is the Milo project listed as one of the official implementations from OPC Foundation? I had a look at the OPC website and the existing Java implementation that they have is legacy and they propose to use the .NET SDK. So is this Milo project an officially supported implementation?

kevinherron, 2019-04-09:
It's not officially supported by the foundation. They've decided to discontinue their support for developing anything but the .NET standard stack and leave it to the existing open source and commercial solutions.

joesan, 2019-04-09:
That's bad! Is this because of Microsoft's heavy involvement in the project?

kevinherron, 2019-04-09:
The reference implementation of the OPC UA spec has always been the .NET version. Microsoft got involved and ported it to .NET Standard and now does much of the development for that version.

The other versions were being sporadically updated by companies contracted by the OPC foundation (Prosys for Java stack, Softing for the C stack), and I think they got tired of having to pay and maintain these somewhat out of date stacks.

I'm not sure any of it is really Microsoft's fault - they're actually pretty supportive and involved with OPC UA.

In the long run this is better because neither of the legacy Java or C stacks were ready or suitable to be used in production.

joesan, 2019-04-09:
So what's the goal with Milo then? Will it have long term support? Any plans to use it officially as part of the supported packages from OPC?

kevinherron, 2019-04-09:
I doubt it. Only the .NET Standard stack will be maintained by them going forward.

Being maintained by the foundation isn't worth much. The "gold standard" UA implementation in C++ by Unified Automation / Ascolab is not maintained by the foundation either. Same with the other commercial stacks.

closed 2019-04-10 by kevinherron

mojtaba-asadollahzadeh, 2019-12-11:
if you were to compare Milo with foundation's sdk which one do you think  is better and why ?

kevinherron, 2019-12-11:
Milo is better. The OPC Foundation’s Java stack is no longer supported or maintained.