eclipse-milo/milo #450: Documentation on how to run Milo examples

joesan, 2019-04-09:
I wanted to try out Milo and I came across the examples folder. While I managed to get the server up and running, I wanted to connect the client and I guess the client-examples pom.xml does not contain a main class definition. I will try to fix that, but can we improve the documentation on adding some instructions about how to run the client and server samples? I can provide a PR!

kevinherron, 2019-04-09:
The client examples are meant to be run out of an IDE right now.

joesan, 2019-04-09:
I just figured that out! So running the client examples will indeed start a server first. I just figured that out!

kevinherron, 2019-04-09:
Yes they will.

On the mailing list yesterday someone started a discussion about documentation if you're interested. Since you're new to the project your take on what kind of documentation would help when getting started would be useful.

joesan, 2019-04-09:
As a starter, I guess adding a README page to the examples might help!

joesan, 2019-04-10:
For the time being, issue is resolved using this PR: https://github.com/eclipse/milo/pull/456

closed 2019-04-10 by joesan

Enomine88, 2019-04-23:
Hi,

as i am new to milo, too: Maybe you would get good docu-clues from me.
But as i am new to github and maven, too, maybe you interpret it as spam ;)

Thanks

Enomine

Enomine88, 2019-04-23:
Do you want to create a Luser-save (Layer 8 Problem) documentation or a documentation for experienced users?

Thanks

Enomine