eclipse-milo/milo #457: Add a CI for all master commits

joesan, 2019-04-10:
Using some CI tool, we could add some auto builds - For a starting point, we could just report on the code test coverage metrics from a CI build. We could also integrate codacy to keep up with the code quality. Travis could be used for this purpose. 

Does this make sense? I can implement this! Let me know!

joesan, 2019-04-10:
Ok I just realized that snapshots are pushed via a mvn deploy command and profiles as I can see it from the pom.xml! So please let me know if this issue makes any sense!

kevinherron, 2019-04-10:
There is already CI in place. Commits to master are tested and deployed, PRs are tested and the status is reported on the PR. Not interested in code coverage right now, especially because a lot of tests are integration tests that cover multiple modules and the coverage tools have a hard time measuring that correctly.

https://ci.eclipse.org/milo

closed 2019-04-10 by kevinherron

joesan, 2019-04-10:
Good to know! I was unaware that Eclipse had a central Jenkins CI server! Where is the Jenkinsfile located?

kevinherron, 2019-04-10:
The jobs are configured on Jenkins rather than by a file in this repo like with Travis CI.

I was previously using Travis CI but it was much slower and less reliable.