eclipse-milo/milo #458: Gitter Channel for the project

joesan, 2019-04-12:
Is there a gitter channel for the project? Could you please set one up? I have few questions on some code bits and I think it would be a good idea to clarify such questions on the channel!

kevinherron, 2019-04-12:
There was but I closed it because I want all questions to be handled on the mailing list. I don’t have time to support the gitter chat as well.

joesan, 2019-04-12:
I have a question on some code. Should I address it via the user group mailing list?

kevinherron, 2019-04-12:
Yes, please use the mailing list for questions. I’d like to focus the community there.

closed 2019-04-12 by kevinherron

heywiki, 2020-06-29:
Is the closed gitter channel archived somewhere? We linked some of our release notes and documentation to issues described there...

kevinherron, 2020-06-29:
Not that I'm aware of. I don't know what gitter or GitLab did with closed channels.