eclipse-milo/milo #459: bad_WriteNoSupported

lizhehong, 2019-04-16:


lizhehong, 2019-04-16:
hello i find this problem:

private static DataValue val(Object o) {
	Variant value = new Variant(o);
	StatusCode status = StatusCode.GOOD;
	return new DataValue(value, status);
}

StatusCode code = client.writeValue(NodeId.parse("ns=2;s=channel2.device2.1"), val(value)).get();

java.lang.AssertionError: StatusCode{name=Bad_WriteNotSupported, value=0x80730000, quality=bad}
	at org.junit.Assert.fail(Assert.java:88)
	at org.junit.Assert.assertTrue(Assert.java:41)
	at cn.hy.iot.OpcUa.AppTest.write(AppTest.java:163)
	at cn.hy.iot.OpcUa.AppTest.test(AppTest.java:130)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:498)
	at org.junit.runners.model.FrameworkMethod$1.runReflectiveCall(FrameworkMethod.java:50)
	at org.junit.internal.runners.model.ReflectiveCallable.run(ReflectiveCallable.java:12)
	at org.junit.runners.model.FrameworkMethod.invokeExplosively(FrameworkMethod.java:47)
	at org.junit.internal.runners.statements.InvokeMethod.evaluate(InvokeMethod.java:17)
	at org.junit.runners.ParentRunner.runLeaf(ParentRunner.java:325)
	at org.junit.runners.BlockJUnit4ClassRunner.runChild(BlockJUnit4ClassRunner.java:78)
	at org.junit.runners.BlockJUnit4ClassRunner.runChild(BlockJUnit4ClassRunner.java:57)
	at org.junit.runners.ParentRunner$3.run(ParentRunner.java:290)
	at org.junit.runners.ParentRunner$1.schedule(ParentRunner.java:71)
	at org.junit.runners.ParentRunner.runChildren(ParentRunner.java:288)
	at org.junit.runners.ParentRunner.access$000(ParentRunner.java:58)
	at org.junit.runners.ParentRunner$2.evaluate(ParentRunner.java:268)
	at org.junit.runners.ParentRunner.run(ParentRunner.java:363)
	at org.eclipse.jdt.internal.junit4.runner.JUnit4TestReference.run(JUnit4TestReference.java:86)
	at org.eclipse.jdt.internal.junit.runner.TestExecution.run(TestExecution.java:38)
	at org.eclipse.jdt.internal.junit.runner.RemoteTestRunner.runTests(RemoteTestRunner.java:538)
	at org.eclipse.jdt.internal.junit.runner.RemoteTestRunner.runTests(RemoteTestRunner.java:760)
	at org.eclipse.jdt.internal.junit.runner.RemoteTestRunner.run(RemoteTestRunner.java:460)
	at org.eclipse.jdt.internal.junit.runner.RemoteTestRunner.main(RemoteTestRunner.java:206)

kevinherron, 2019-04-16:
Most servers don't support writing a StatusCode or timestamp with a value. Try creating your value without them:
```java
DataValue dv = new DataValue(v, null, null);
```

If that doesn't work then your server just doesn't support writing to the value you're trying to write to for some reason.

lizhehong, 2019-04-16:
@kevinherron but i install other scada software,it has opc ua server .
1.when i use milo write value to it ,it can response ok and change server value.this software is ignition
2.i use opc ua client ex: prosys-opc-ua-client . this client can write kepware opc ua service tag value
3.i use prosys-opc-ua-client  java sdk write value to kepware.this client can write 

so i do not koow ,milo not write value to kepware???

thank you !!!

kevinherron, 2019-04-16:
The new error `Bad_TypeMismatch` means you're writing the wrong kind of value now.

OPC UA requires strict type checking of values being written. Do you know what the datatype is?

lizhehong, 2019-04-16:
my code 
> write(client,(short)12);
kepware tag value type : Word

lizhehong, 2019-04-16:
my code

`write(client,(short)12);`

kepware tag value type : Word

kevinherron, 2019-04-16:
What's the OPC UA datatype? If it's `UInt16`, then you'd have to write like this:

```java
write(client, ushort(12));
```

There are `UByte`, `UShort`, `UInteger`, and `ULong` types provided by Milo that correspond to unsigned OPC UA datatypes.

lizhehong, 2019-04-16:
thank you very much . i konw hot to use milo ,milo must use milo data type ,because milo encoder data from these milo data type

closed 2019-04-17 by kevinherron