eclipse-milo/milo #460: connect to Siemens PLC

tandi960505, 2019-04-26:
How does the OPCUA-SERVER implemented by Eclipse milo connect to Siemens PLC?

closed 2019-04-26 by tandi960505

kevinherron, 2019-04-26:
It doesn't, but if you have an S7-1500 with a built-in OPC UA server you could use Milo as an OPC UA client to connect to it.

This is an SDK for developing OPC UA clients or servers, not an OPC UA server product like Kepware with a driver for Siemens PLCs.

agawande72, 2020-08-13:
Do we have some limitation in Eclipse milo library ? 
Due to which, Eclipse Milo OPCUA server cannot connects to Siemens PLC OPCUA server.

Example:
I have developed OPCUA Server using Eclipse Milo Library. Currently this server connects to all other OPCUA servers excepts Siemens PLC OPCUA server.

Any specific reason why we cannot connect to Siemens PLC. 
Any Suggestion, If we want to add support to our OPCUA server to connect to Siemens PLC OPCUA Server

Thanks in advance

kevinherron, 2020-08-13:
@agawande72 no limitations, this library is used to connect to the Siemens OPC UA server by many people.

Open a new ticket with additional information and logs.