eclipse-milo/milo #463: milo-0.2.5，about account login problem

tandi960505, 2019-04-30:
opcua server code:
``` java
        // account validator
        UsernameIdentityValidator identityValidator = new UsernameIdentityValidator(
            true,
            authChallenge -> {
                String username = authChallenge.getUsername();
                String password = authChallenge.getPassword();

                boolean userOk = "user".equals(username) && "123456".equals(password);
                boolean adminOk = "admin".equals(username) && "123456".equals(password);

                return userOk || adminOk;
            }
        );
     ......
OpcUaServerConfig serverConfig = OpcUaServerConfig.builder()
     ......
            .setIdentityValidator(new CompositeValidator(identityValidator, x509IdentityValidator))
            .setSecurityPolicies(
                EnumSet.of(
                    SecurityPolicy.Basic128Rsa15,
                    SecurityPolicy.Basic256,
                    SecurityPolicy.Basic256Sha256,
                    SecurityPolicy.Aes128_Sha256_RsaOaep,
                    SecurityPolicy.Aes256_Sha256_RsaPss))
            .setUserTokenPolicies(ImmutableList.of(USER_TOKEN_POLICY_USERNAME))
            .build();
```

opcua client code:
``` java
public interface ClientExample {
    default String getEndpointUrl() {
        return "opc.tcp://127.0.0.1:8874";
    }

    // security policy
    default SecurityPolicy getSecurityPolicy() {
        return SecurityPolicy.Basic128Rsa15;
    }

    // account
    default IdentityProvider getIdentityProvider() {
        return new UsernameProvider("user", "123456");
    }
    void run(OpcUaClient client, CompletableFuture<OpcUaClient> future) throws Exception;
}
```

exception：
```
09:31:49.119 [ua-netty-event-loop-5] ERROR o.e.m.o.s.c.h.UaTcpClientMessageHandler - [remote=/127.0.0.1:8874] errorMessage=ErrorMessage{error=StatusCode{name=Bad_SecurityChecksFailed, value=0x80130000, quality=bad}, reason=status=Bad_SecurityChecksFailed, description=An error occurred verifying security.}
09:31:49.121 [ua-netty-event-loop-6] ERROR o.e.m.o.s.s.h.UaTcpServerAsymmetricHandler - [remote=/127.0.0.1:60569] Exception caught; sent ErrorMessage{error=StatusCode{name=Bad_SecurityChecksFailed, value=0x80130000, quality=bad}, reason=status=Bad_SecurityChecksFailed, description=An error occurred verifying security.}
io.netty.handler.codec.DecoderException: UaException: status=Bad_SecurityChecksFailed, message=An error occurred verifying security.
	at io.netty.handler.codec.ByteToMessageDecoder.callDecode(ByteToMessageDecoder.java:461)
	at io.netty.handler.codec.ByteToMessageDecoder.channelRead(ByteToMessageDecoder.java:267)
	at io.netty.channel.AbstractChannelHandlerContext.invokeChannelRead(AbstractChannelHandlerContext.java:356)
	at io.netty.channel.AbstractChannelHandlerContext.invokeChannelRead(AbstractChannelHandlerContext.java:342)
	at io.netty.channel.AbstractChannelHandlerContext.fireChannelRead(AbstractChannelHandlerContext.java:335)
	at io.netty.channel.DefaultChannelPipeline$HeadContext.channelRead(DefaultChannelPipeline.java:1304)
	at io.netty.channel.AbstractChannelHandlerContext.invokeChannelRead(AbstractChannelHandlerContext.java:356)
	at io.netty.channel.AbstractChannelHandlerContext.invokeChannelRead(AbstractChannelHandlerContext.java:342)
	at io.netty.channel.DefaultChannelPipeline.fireChannelRead(DefaultChannelPipeline.java:921)
	at io.netty.channel.nio.AbstractNioByteChannel$NioByteUnsafe.read(AbstractNioByteChannel.java:135)
	at io.netty.channel.nio.NioEventLoop.processSelectedKey(NioEventLoop.java:646)
	at io.netty.channel.nio.NioEventLoop.processSelectedKeysOptimized(NioEventLoop.java:581)
	at io.netty.channel.nio.NioEventLoop.processSelectedKeys(NioEventLoop.java:498)
	at io.netty.channel.nio.NioEventLoop.run(NioEventLoop.java:460)
	at io.netty.util.concurrent.SingleThreadEventExecutor$2.run(SingleThreadEventExecutor.java:131)
[... 45 lines cut ...]
	at org.eclipse.milo.examples.client.ReadExample.main(ReadExample.java:34)
Caused by: org.eclipse.milo.opcua.stack.core.UaException: status=Bad_SecurityChecksFailed, description=An error occurred verifying security.
	at org.eclipse.milo.opcua.stack.client.handlers.UaTcpClientMessageHandler.onError(UaTcpClientMessageHandler.java:716)
	at org.eclipse.milo.opcua.stack.client.handlers.UaTcpClientMessageHandler.decodeMessage(UaTcpClientMessageHandler.java:411)
	at org.eclipse.milo.opcua.stack.client.handlers.UaTcpClientMessageHandler.decode(UaTcpClientMessageHandler.java:392)
	at io.netty.handler.codec.ByteToMessageCodec$1.decode(ByteToMessageCodec.java:42)
	at io.netty.handler.codec.ByteToMessageDecoder.decodeRemovalReentryProtection(ByteToMessageDecoder.java:491)
	at io.netty.handler.codec.ByteToMessageDecoder.callDecode(ByteToMessageDecoder.java:430)
	at io.netty.handler.codec.ByteToMessageDecoder.channelRead(ByteToMessageDecoder.java:267)
	at io.netty.handler.codec.ByteToMessageCodec.channelRead(ByteToMessageCodec.java:103)
	at io.netty.channel.AbstractChannelHandlerContext.invokeChannelRead(AbstractChannelHandlerContext.java:356)
	at io.netty.channel.AbstractChannelHandlerContext.invokeChannelRead(AbstractChannelHandlerContext.java:342)
	at io.netty.channel.AbstractChannelHandlerContext.fireChannelRead(AbstractChannelHandlerContext.java:335)
	at io.netty.channel.DefaultChannelPipeline$HeadContext.channelRead(DefaultChannelPipeline.java:1304)
	at io.netty.channel.AbstractChannelHandlerContext.invokeChannelRead(AbstractChannelHandlerContext.java:356)
	at io.netty.channel.AbstractChannelHandlerContext.invokeChannelRead(AbstractChannelHandlerContext.java:342)
	at io.netty.channel.DefaultChannelPipeline.fireChannelRead(DefaultChannelPipeline.java:921)
	at io.netty.channel.nio.AbstractNioByteChannel$NioByteUnsafe.read(AbstractNioByteChannel.java:135)
	at io.netty.channel.nio.NioEventLoop.processSelectedKey(NioEventLoop.java:646)
	at io.netty.channel.nio.NioEventLoop.processSelectedKeysOptimized(NioEventLoop.java:581)
	at io.netty.channel.nio.NioEventLoop.processSelectedKeys(NioEventLoop.java:498)
	at io.netty.channel.nio.NioEventLoop.run(NioEventLoop.java:460)
	at io.netty.util.concurrent.SingleThreadEventExecutor$2.run(SingleThreadEventExecutor.java:131)
	at java.lang.Thread.run(Thread.java:748)
```

kevinherron, 2019-04-30:
This is probably failing because you enabled security and now you have to mutually trust the client and server certificates. When the example starts it should log the security directory where you can go and move the certificates into the corresponding trusted folders.

tandi960505, 2019-04-30:
> 这可能是失败的，因为您启用了安全性，现在您必须相互信任客户端和服务器证书。示例启动时，应记录安全目录，然后将证书移动到相应的受信任文件夹中。

thank！

tandi960505, 2019-04-30:
> This is probably failing because you enabled security and now you have to mutually trust the client and server certificates. When the example starts it should log the security directory where you can go and move the certificates into the corresponding trusted folders.

thank!

closed 2019-04-30 by tandi960505