eclipse-milo/milo #470: Question: VariableNode NodeId 0.2.5 to 0.3.0

deify123, 2019-05-10:
Hello,

until yesterday I used Milo 0.2.5 and when I created an ObjectType with Variables it was created as 'NodeId.Variable' but after I updated to 0.3.0-M1 it is created as 'NodeId/(nsId):Variable', can this behavior be changed using parameters to how it was before or is it just hard-coded this way now?

Best Regards

kevinherron, 2019-05-10:
It looks like you can change the behavior by creating your own NodeFactory
subclass that overrides the instanceNodeId() method.

deify123, 2019-05-13:
That worked perfectly thanks.

For future reference: I also had to add MyNodeFactory as a field and initialize it in the Namespace where I needed it and override getNodeFactory to use the new one too.

closed 2019-05-13 by deify123